gotthardp/lorawan-server #621: Parse uplink with variable frames length

elisascheer, 2019-04-04:
Hi,

I've got a device which sends frames with variable length (I can't know in advance the frame's length).
For example device can send :

`Header n (1 byte) | Data n x bytes | Header n+1 (1 byte) | data n+1 x bytes | ... | ...`

so the header is giving information about the length of followed data and so on...

So I was wondering if it was possible to implement such a mechanism in the parse uplink ?

best regards

gotthardp, 2019-04-04:
Hello. Yes, it is possible.

elisascheer, 2019-04-04:
Thanks,

how to implement the simplest way ?

I was thinking about doing :

`lists:foreach( fun(X) -> 
  %Processing elements
end, binary_to_list(Data))`

gotthardp, 2019-04-04:
If the variable length is just options, then you can match these options in the fun header:
```erlang
fun (Fields, <<one frame>>) ->
        Fields#{whatever1};
    (Fields, <<another frame>>) ->
        Fields#{whatever2}
end.
```
If you have multiple variable sized fields, then you need to define a recursive function, which requires a funny trick:
```erlang
fun (Fields, Bin) ->
    Parse =
        fun (P, F, <<1, Rest/binary>>) ->
                P(P, F#{first => 1}, Rest);
            (P, F, <<2, 3, Rest/binary>>) ->
                P(P, F#{second => 23}, Rest);
            (_P, F, <<>>) ->
                F
        end,
    Parse(Parse, Fields, Bin)
end.
```
Recursion needs a named function like explained in many [recursion tutorials](https://learnyousomeerlang.com/recursion). To achieve this, a fun Parse is defined, which takes itself as a first argument (P), so it can be called recursively. The second argument (F) are the fields that get amended as the fields get parsed and the last argument is the binary, which gets shorter and shorter.

elisascheer, 2019-04-05:
Hi again, 
Oh ok I see with the recursive fun trick. 
Is that require to only use the `fun` function inside the Parse uplink?

Thanks

gotthardp, 2019-04-05:
Yes, the current implementation requires the Parse uplink to include a single `fun`. I know it is limiting for complex data structures, but that;s the current situation.

elisascheer, 2019-04-11:
Hi again,

So I tried to implement the recursive function as you explained me by doing :  

```
fun (Fields, <<Code, Packet/binary>>) ->
    Parse =
          fun (P, F, <<Length, Value:Length/bytes, Rest/binary>>) ->
                P(P, F#{ length => Length, value => Value}, Rest);
                (_P, F, <<>>) ->
                F
          end(Packet),
    Parse(Parse, Fields, Packet)
end.
```

There is no error when I created the handler but when a frame is received, JSON data is `{}` empty because there was a `parse_failed` (without args)

elisascheer, 2019-04-11:
For information I tested it by sending this frame : 
`400212450150` so I was expected to see at least output as `{"length" : 2, "value" : 18 }`

gotthardp, 2019-04-11:
There is a strange `(Packet)` after the `end`. This works:
```erlang
fun (Fields, <<Code, Packet/binary>>) ->
    Parse =
          fun (P, F, <<Length, Value:Length/bytes, Rest/binary>>) ->
                P(P, F#{ length => Length, value => Value}, Rest);
                (_P, F, <<>>) ->
                F
          end,
    Parse(Parse, Fields, Packet)
end.
```

elisascheer, 2019-04-11:
Thanks for answering,

I tested but it's always the same error and it returns an empty JSON in the MQTT topic.

gotthardp, 2019-04-11:
Is there something more in the logs?

elisascheer, 2019-04-11:
logs says :  
```parse_failed  {error,function_clause}```

I tried by removing the ``` fun (Fields, <<Code, Packet/binary>>)``` by ``` fun (Fields, Packet)``` and it returns json now but with an offset of 1 byte

gotthardp, 2019-04-11:
What offset?

elisascheer, 2019-04-11:
By offset I mean that Code becomes Length as I removed it and replaced by ```Packet```. 
I retried with another device still sending 40 02 12 45 01 50 frame but the output is : ``` {"length":1,"value":"P"} ```
using the following parser uplink :  
```
fun (Fields, <<Code, Packet/binary>>) ->
    Parse =
          fun (P, F, <<Length, Value:Length/binary, Rest/binary>>) ->
                P(P, F#{ length => Length, value => Value}, Rest);
                (_P, F, <<>>) ->
                F
          end,
    Parse(Parse, Fields, Packet)
end.

```

gotthardp, 2019-04-11:
And `{"length":1,"value":"P"}` is what you should get. So it is working fine :)

elisascheer, 2019-04-11:
oh okay I got it : length for 01 and P for 50, is taht normal that it is ASCII text? I was expecting to get 80 as value (or maybe the field ``` Value:Length/binary ``` converts it to ASCII?)

gotthardp, 2019-04-11:
By saying `Value:Length/binary` you say you want a binary value, which is by JSON converted to string. If you want integer do `Value:Length`.

elisascheer, 2019-04-11:
Okay I got it.
If ``` Value:Length ``` and Length equals 1 then it's just taking one bit of Value and not one byte isn't it?

gotthardp, 2019-04-11:
Oh yeah, sorry. If you want the length in bytes, then it's `Value:Length/unit:8`.
The magic is explained here http://erlang.org/doc/programming_examples/bit_syntax.html#segments

closed 2019-05-27 by elisascheer