gotthardp/lorawan-server #622: latest commint : failed at compilation

pdl86, 2019-04-09:
Hello 

I have tried to compile for Mlinux the latest committed version ( hence the reference 0.6.71) , but it failed: 

withe a missing module : node_modules/leaflet/dist/leaflet.js

we have used imyller node  git hub repository ( node and node-contrib).
Note that mlinux ( multitech) is "stuck" at node version 0.10.47  as the hardware is not supporting 'floating point'.

Could it be the reason ? 

regards

extract of the bitbake trace 

uffer@3.6.0, browserify-zlib@0.1.4, crypto-browserify@3.2.8)
| âââ ng-file-upload@7.3.9
| âââ codemirror@5.45.0
| âââ restangular@1.5.2 (lodash@4.17.11)
| cp: cannot stat ânode_modules/leaflet/dist/leaflet.jsâ: No such file or directory
| ===> Hook for compile failed!
|
| Makefile:35: recipe for target 'release' failed
| make: *** [release] Error 1
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/transnumeric/mlinux-4.1.6/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.71-r0/temp/log.do_compile.30329)
ERROR: Task (/home/transnumeric/mlinux-4.1.6/layers/meta-lorawan/recipes-core/lorawan-server/lorawan-server_0.6.71.bb:do_compile) failed with exit code '1'

pdl86, 2019-04-09:
What's strange is tthe the "missing" file generating the error seems to be there: 
/home/transnumeric/mlinux-4.1.6/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.71-r0/git/node_modules/ui-leaflet/node_modules/leaflet/dist/leaflet.js

hallard, 2019-04-09:
@pdl86 

I installed the leaflet module on host (not sure it's the reason but I know lorawan-server needs nodejs to build, so may be that's the reason
````
$ sudo npm install -g leaflet
/usr/lib
└── leaflet@1.4.0
````

Could you try again?

gotthardp, 2019-04-11:
I cannot reproduce this. It works for me. Could you please re-try with the latest meta-lorawan updates?

pdl86, 2019-04-12:
In fact I did an mistake and recompile 0.6.7 and not the latest commit.
DISREGARD previous message ( suppressed)

Then on the lastest commit 
https://github.com/gotthardp/lorawan-server/commit/6ecdda07b1d438b1da65cd19fc08da783a5569b8

I have still the same error
| âââ ng-file-upload@7.3.9
| âââ codemirror@5.45.0
| âââ restangular@1.5.2 (lodash@4.17.11)
| cp: cannot stat ânode_modules/leaflet/dist/leaflet.jsâ: No such file or directory
| ===> Hook for compile failed!
|
| Makefile:35: recipe for target 'release' failed
| make: *** [release] Error 1
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/transnumeric/mlinux-4.1.6/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.71-r0/temp/log.do_compile.9160)

node_modules/leaflet/dist/leaflet.js   missing , but where ??

I confirm that the error appear on THIS commit, as I'm able to compile correctly 0.6.7 

regards
Philippe

hallard, 2019-04-12:
I cloned the latest meta-lorawan then I copied file 0.6.7. bb to 0.6.71.bb and changed to latest commit of lorawan-server 

Same issue, here is the log trace

```
Initialising tasks: 100% |####################################################################################################################| Time: 0:00:02
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: lorawan-server-0.6.71-r0 do_compile: oe_runmake failed
ERROR: lorawan-server-0.6.71-r0 do_compile: Function failed: do_compile (log file is located at /home/transnumeric/mlinux-4.1.6/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.71-r0/temp/log.do_compile.13269)
ERROR: Logfile of failure stored in: /home/transnumeric/mlinux-4.1.6/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.71-r0/temp/log.do_compile.13269
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4 release
| ===> Verifying dependencies...
| ===> Compiling cowlib
| ===> Compiling recon
| ===> Compiling rabbit_common
| _build/default/lib/rabbit_common/src/ssl_compat.erl:47: Warning: ssl:connection_info/1: removed in 20.0; use ssl:connection_information/[1,2] instead
| _build/default/lib/rabbit_common/src/ssl_compat.erl:67: Warning: ssl:connection_info/1: removed in 20.0; use ssl:connection_information/[1,2] instead
|
| _build/default/lib/rabbit_common/src/delegate.erl:245: Warning: erlang:get_stacktrace/0: deprecated; use the new try/catch syntax for retrieving the stack backtrace
|
| _build/default/lib/rabbit_common/src/vm_memory_monitor.erl:309: Warning: erlang:get_stacktrace/0: deprecated; use the new try/catch syntax for retrieving the stack backtrace
|
| ===> Compiling amqp_client
| ===> Compiling eid
| ===> Compiling cbor
| ===> Compiling iso8601
| ===> Compiling ranch
| ===> Compiling gen_smtp
| _build/default/lib/gen_smtp/src/socket.erl:124: Warning: ssl:ssl_accept/1: deprecated; use ssl:handshake/1 instead
| _build/default/lib/gen_smtp/src/socket.erl:233: Warning: ssl:ssl_accept/3: deprecated; use ssl:handshake/3 instead
|
| _build/default/lib/gen_smtp/src/smtp_util.erl:71: Warning: crypto:rand_uniform/2 is deprecated and will be removed in a future release; use rand:uniform/1
| _build/default/lib/gen_smtp/src/smtp_util.erl:71: Warning: crypto:rand_uniform/2 is deprecated and will be removed in a future release; use rand:uniform/1
|
| ===> Compiling erlmongo
| ===> Compiling cowboy
| ===> Compiling gun
| ===> Compiling jsx
| ===> Compiling goldrush
| ===> Compiling lager
| ===> Compiling emqttc
| _build/default/lib/emqttc/src/emqttc_socket.erl:204: Warning: gen_fsm:send_all_state_event/2 is deprecated and will be removed in a future release; use gen_statem:cast/2
| _build/default/lib/emqttc/src/emqttc_socket.erl:215: Warning: gen_fsm:send_all_state_event/2 is deprecated and will be removed in a future release; use gen_statem:cast/2
| _build/default/lib/emqttc/src/emqttc_socket.erl:233: Warning: gen_fsm:send_event/2 is deprecated and will be removed in a future release; use gen_statem:cast/2
| _build/default/lib/emqttc/src/emqttc_socket.erl:242: Warning: gen_fsm:send_all_state_event/2 is deprecated and will be removed in a future release; use gen_statem:cast/2
|
| _build/default/lib/emqttc/src/emqttc.erl:168: Warning: gen_fsm:start_link/3 is deprecated and will be removed in a future release; use gen_statem:start_link/3
| _build/default/lib/emqttc/src/emqttc.erl:191: Warning: gen_fsm:start_link/4 is deprecated and will be removed in a future release; use gen_statem:start_link/4
| _build/default/lib/emqttc/src/emqttc.erl:199: Warning: gen_fsm:sync_send_all_state_event/2 is deprecated and will be removed in a future release; use gen_statem:call/2
| _build/default/lib/emqttc/src/emqttc.erl:252: Warning: gen_fsm:send_event/2 is deprecated and will be removed in a future release; use gen_statem:cast/2
| _build/default/lib/emqttc/src/emqttc.erl:260: Warning: gen_fsm:sync_send_event/2 is deprecated and will be removed in a future release; use gen_statem:call/2
| _build/default/lib/emqttc/src/emqttc.erl:333: Warning: gen_fsm:send_event/2 is deprecated and will be removed in a future release; use gen_statem:cast/2
| _build/default/lib/emqttc/src/emqttc.erl:336: Warning: gen_fsm:sync_send_event/3 is deprecated and will be removed in a future release; use gen_statem:call/3
| _build/default/lib/emqttc/src/emqttc.erl:348: Warning: gen_fsm:send_event/2 is deprecated and will be removed in a future release; use gen_statem:cast/2
| _build/default/lib/emqttc/src/emqttc.erl:356: Warning: gen_fsm:sync_send_event/3 is deprecated and will be removed in a future release; use gen_statem:call/3
| _build/default/lib/emqttc/src/emqttc.erl:364: Warning: gen_fsm:send_event/2 is deprecated and will be removed in a future release; use gen_statem:cast/2
| _build/default/lib/emqttc/src/emqttc.erl:477: Warning: gen_fsm:cancel_timer/1 is deprecated and will be removed in a future release; use erlang:cancel_timer/1
| _build/default/lib/emqttc/src/emqttc.erl:494: Warning: gen_fsm:send_event/2 is deprecated and will be removed in a future release; use gen_statem:cast/2
| _build/default/lib/emqttc/src/emqttc.erl:646: Warning: variable 'Name' is unused
| _build/default/lib/emqttc/src/emqttc.erl:758: Warning: gen_fsm:cancel_timer/1 is deprecated and will be removed in a future release; use erlang:cancel_timer/1
[... cut ...]
ERROR: Task (/home/transnumeric/mlinux-4.1.6/layers/meta-lorawan/recipes-core/lorawan-server/lorawan-server_0.6.71.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1096 tasks of which 1094 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/transnumeric/mlinux-4.1.6/layers/meta-lorawan/recipes-core/lorawan-server/lorawan-server_0.6.71.bb:do_compile
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

hallard, 2019-04-12:
Ok, got it was searching for package `leaflet` at compile time (not only `ui-leaflet`) I installed it manually into tmp `build/git/node_modules` folder and saved my life (even if `leaflet` is contained into `ui-leaflet`)

so `package.json` of this repo should contains `"leaflet": "0.7.7"` for compile to works (I did not had this problem compiling for LorixOne, just for Multitech)

```json
{
  "private": true,
  "dependencies": {
    "jquery": ">=1.0",
    "ng-admin": ">=1.0.7",
    "angular-simple-logger": ">=0.0.1",
    "angular-google-chart": "1.0.0-beta.1",
    "ui-leaflet": "1.0.3",
    "leaflet": "0.7.7",
    "vis": "4.20.1",
    "angular-visjs": "4.16.0",
    "angular-bootstrap-colorpicker": "v3.0.32"
  }
}
```

gotthardp, 2019-04-12:
@hallard, good catch! I never had this trouble, but adding `leaflet` doesn't harm anything. I just made the fix.

hallard, 2019-04-15:
@gotthardp 
works fine, thank for fixing this one

closed 2019-04-15 by gotthardp