gotthardp/lorawan-server #623: 0.6.7 packing issue

pdl86, 2019-04-09:
When compiling / packaing lorawan-server for mlinux Multitehc conduit, I've got the following warning 

WARNING: lorawan-server-0.6.7-r0 do_package: QA Issue: lorawan-server: Files/directories were installed but not shipped in any package:
  /etc
  /etc/default
  /etc/init.d
  /etc/default/lorawan-server
  /etc/init.d/lorawan-server
Please set FILES such that these items are packaged. Alternatively if they are unneeded, avoid installing them or delete them within do_install.
lorawan-server: 5 installed and not shipped files. [installed-vs-shipped]

they are missing in .../mlinux-4.1.6/build/tmp/work/arm926ejste-mlinux-linux-gnueabi/lorawan-server/0.6.7-r0/

while being present in .../mlinux-4.1.6/layers/meta-lorawan/recipe-core/lorawan-server/files 

thanks for correcting

hallard, 2019-04-09:
@pdl86 looks like you got the same as I have on Lorix One with Multitech
Quick fix is following
https://github.com/gotthardp/meta-lorawan/issues/6

gotthardp, 2019-04-11:
The gotthardp/meta-lorawan#6 was just fixed.

closed 2019-04-11 by gotthardp