gotthardp/lorawan-server #624: bag in version 0.6.6

dencorpz1, 2019-04-10:
Good day!
today saw such a problem.
When receiving data from 2 base stations, for some reason, the base base station has become a base with poor reception conditions, and when sending a packet, the server sends a packet through this base station. Log packages in the attachment. How can I fix this?

dencorpz1, 2019-04-10:
[lora_server_bag.zip](https://github.com/gotthardp/lorawan-server/files/3062642/lora_server_bag.zip)

gotthardp, 2019-04-10:
Hello. What exactly is the problem? The server sends packets through the gateway with the best reception.

dencorpz1, 2019-04-11:
Good day!
The fact is that this base station is located 1 kilometer from the device, and the second base station is 200 meters away. The level of SNR shows that the second base station SNR is much better. The problem is that the package does not arrive from the first base station to the device, and it is impossible to interact with the device. How can I select a base station to send a packet.

gotthardp, 2019-04-11:
Yeah, some RF interference could decrease the RSSI. Any lora frame may get lost, so if your device expects a response you should use confrmed frames.
What you describe is not a bug; it's the current behaviour. Manual selection of a gateway is not implemented at this moment.

ruipalma, 2020-05-20:
I have exactly the same problem, it will be really usefull to have a "lock to gateway" in device config.
Maybe the SNR is a better option to choose the best GW.