gotthardp/lorawan-server #625: Debugging Erlang parse function

nupik, 2019-04-10:
Hello! It will be very helpful if possible to try parse function with online compiler. For example: https://rextester.com/l/erlang/
Could you provide example code for it, where I can enter payload and see JSON result.

gotthardp, 2019-04-11:
Hello. You can always write in the https://rextester.com/l/erlang/ something like this
```erlang
%Erlang 7.3
%dont change first 3 lines

-module(source).
-export([entry_point/0]).

entry_point() ->
    Parse =
        fun (Fields, <<16#0402:16, Temp:16/signed>>) ->
                Fields#{temp => Temp};
            (Fields, <<16#0405:16, Level>>) ->
                Fields#{level => Level}
        end,

    io:write(
        Parse(#{}, <<04, 02, 00, 01>>)).
```

This takes the binary and prints the map, which is amlost the final JSON.

nupik, 2019-04-11:
Thank you very much! This is that I need!

closed 2019-04-11 by nupik