gotthardp/lorawan-server #626: Set up/down frame counters

stefanobimbo, 2019-04-12:
I am running v 0.6.6

I use Telit RE866A-1 as end-node. This device doesn't allow me to freely set values of up and down counters on end-node side. Sometimes happens that network server and end-node counters don't match. In this case, I'm forced to manually change counters values through web interface, to restore correct behavior.
I would like to change frame counters values automatically using my application in the following way: 
- frame counter values read on the end-node side, are sent to the application in the uplink payload 
- the application checks if they are different from the network server counterparts
- if counters values are different, application calls the API "/api/nodes/<DevAddr>" for a PUT to reset values using the end-node values

 I know the logic is not fantastic, but I have no other chance (I think). I hope frame counters values don't differ too often (really I hope they will never differ), but who knows?

The problem I see is that when I call the api to put new frame counters values, I also set other device data. Before doing PUT I call GET to read actual device data and then use the just read data (changing only the frame counter values) for the PUT, but I'm sure I will have, soon or later, a race condition overwritting new device data updated by the network server between my GET and following PUT.

Any ideas? Is there a better way to do that? Can you think of a specialized API (or any other way) to update only frame counters values for a given node?

gotthardp, 2019-04-12:
Hello. Since 0.6.7 there is an (undocumented) API to modify a single field. So one could use GET/PUT with `http://192.168.1.1:8080/api/nodes/00001234/fcntdown` do modify just the downlink counter. It works for any field, but with devices and nodes only. Would this be a solution to your issue?

stefanobimbo, 2019-04-14:
Yes, this is a perfect solution for my issue. Thanks

closed 2019-04-15 by gotthardp