gotthardp/lorawan-server #627: Decoding payload enhancement :

hallard, 2019-04-15:
From beginning I'm decoding payload data after this server but I really would like doing it on the server. 
Of course we can do that in erlang payload decoder but to be honest my knowledge of erlang is so poor that it's really difficult to me, also when decoding needs some other treatment or adjustment.
So my question is simple, would it be possible to integrate external decoding engine, such as JS/NodeJS script or other like in thingsboard or TTN ?
Not sure it's easy bit worth asking and would save my life because also lot of decoder are writren in JS.
What are your thought? Even if I have a small idea ;-)

gotthardp, 2019-04-15:
You can send a raw binary out and decode it in your app. That's not what you want?
Or you want to decode it using JSON and then send it to some cloud like Azure?

hallard, 2019-04-17:
Sure, but I want to keep final app connection with the server, like today
```
Sensor <-> gotthardp <-> internal Erlang Decoder (json) <-> gotthardp <-> Final app (WS/MQTT)
```

And I want to keep final app connection with the server, so I mean more something like that
```
Sensor <-> gotthardp <-> (raw/json, ...) External decoder (json) <-> gotthardp <-> Final app (WS/MQTT)
```

Like that I do not need to change anything on final app on new sensors, just add new decoder function in the external one

Does this makes sense?