gotthardp/lorawan-server #631: [Enhancement] Semtech LNS protocol

xypron, 2019-05-02:
Are there any plans to support the Semtech LNS protocol (https://doc.sm.tc/station/tcproto.html)?

gotthardp, 2019-05-02:
So far no plans. The protocol seems to be developed (the sm.tc domain is owned) by Trackio GmbH, so Semtech is not(?) involved. Once this software attracts some users we could implement it.

xypron, 2019-05-02:
This is the the announcement by Semtech
https://www.semtech.com/company/press/semtech-announces-open-source-software-for-lora-based-gateways-at-the-things-conference

It pointed me to https://github.com/lorabasics/basicstation, which in turn links to https://doc.sm.tc/station.

https://github.com/lorabasics/basicstation/blob/master/LICENSE indicates Semtech as copyright owner.

gotthardp, 2019-05-02:
Alright, this changes everything. I will implement that.

gotthardp, 2019-05-31:
Done. Will be in v0.7.0

closed 2019-05-31 by gotthardp