gotthardp/lorawan-server #632: Many errors in logs when using MQTTS

nicoinn, 2019-05-07:
I've configured lorawanserver to connect to my mqtt server (VerneMQ) over mqtts, with a certificate provided by Let's encrypt.

I've added the (the let's encrypt certificate)[https://letsencrypt.org/certs/trustid-x3-root.pem.txt] to my ca-certificates.

It seems to be working fine as I'm getting my messages through, but my logs are polluted with the error messages below.

Any hints at what is going wrong here ?

(I use lorawan server in Docker, using the alpine docker-file provided in this repo) 

```
2019-05-07 12:24:23.059 [debug] <0.408.0>@lorawan_connector_mqtt:handle_info:241 Connector <<"MQTT">> to <<"mqtts://mqtt.innocenti.se:1884">> (connected311) failed: {shutdown,ssl_closed} (count: 0)
2019-05-07 12:24:23.059 [debug] <0.408.0>@lorawan_connector_mqtt:reconnect:141 Reconnecting
2019-05-07 12:24:54.492 [debug] <0.408.0>@lorawan_connector_mqtt:handle_info:241 Connector <<"MQTT">> to <<"mqtts://mqtt.innocenti.se:1884">> (connected311) failed: {shutdown,ssl_closed} (count: 0)
2019-05-07 12:24:54.492 [debug] <0.408.0>@lorawan_connector_mqtt:reconnect:141 Reconnecting
2019-05-07 12:25:24.698 [debug] <0.408.0>@lorawan_connector_mqtt:handle_info:241 Connector <<"MQTT">> to <<"mqtts://mqtt.innocenti.se:1884">> (connected311) failed: {shutdown,ssl_closed} (count: 0)
2019-05-07 12:25:24.698 [debug] <0.408.0>@lorawan_connector_mqtt:reconnect:141 Reconnecting
2019-05-07 12:25:54.927 [debug] <0.408.0>@lorawan_connector_mqtt:handle_info:241 Connector <<"MQTT">> to <<"mqtts://mqtt.innocenti.se:1884">> (connected311) failed: {shutdown,ssl_closed} (count: 0)
2019-05-07 12:25:54.927 [debug] <0.408.0>@lorawan_connector_mqtt:reconnect:141 Reconnecting
2019-05-07 12:26:25.145 [debug] <0.408.0>@lorawan_connector_mqtt:handle_info:241 Connector <<"MQTT">> to <<"mqtts://mqtt.innocenti.se:1884">> (connected311) failed: {shutdown,ssl_closed} (count: 0)
2019-05-07 12:26:25.145 [debug] <0.408.0>@lorawan_connector_mqtt:reconnect:141 Reconnecting
2019-05-07 12:26:55.779 [debug] <0.408.0>@lorawan_connector_mqtt:handle_info:241 Connector <<"MQTT">> to <<"mqtts://mqtt.innocenti.se:1884">> (connected311) failed: {shutdown,ssl_closed} (count: 0)
2019-05-07 12:26:55.779 [debug] <0.408.0>@lorawan_connector_mqtt:reconnect:141 Reconnecting
2019-05-07 12:27:25.931 [debug] <0.408.0>@lorawan_connector_mqtt:handle_info:241 Connector <<"MQTT">> to <<"mqtts://mqtt.innocenti.se:1884">> (connected311) failed: {shutdown,ssl_closed} (count: 0)
2019-05-07 12:27:25.931 [debug] <0.408.0>@lorawan_connector_mqtt:reconnect:141 Reconnecting
2019-05-07 12:27:56.072 [debug] <0.408.0>@lorawan_connector_mqtt:handle_info:241 Connector <<"MQTT">> to <<"mqtts://mqtt.innocenti.se:1884">> (connected311) failed: {shutdown,ssl_closed} (count: 0)
2019-05-07 12:27:56.072 [debug] <0.408.0>@lorawan_connector_mqtt:reconnect:141 Reconnecting
2019-05-07 12:28:26.214 [debug] <0.408.0>@lorawan_connector_mqtt:handle_info:241 Connector <<"MQTT">> to <<"mqtts://mqtt.innocenti.se:1884">> (connected311) failed: {shutdown,ssl_closed} (count: 0)
2019-05-07 12:28:26.214 [debug] <0.408.0>@lorawan_connector_mqtt:reconnect:141 Reconnecting
2019-05-07 12:28:56.357 [debug] <0.408.0>@lorawan_connector_mqtt:handle_info:241 Connector <<"MQTT">> to <<"mqtts://mqtt.innocenti.se:1884">> (connected311) failed: {shutdown,ssl_closed} (count: 0)
2019-05-07 12:28:56.357 [debug] <0.408.0>@lorawan_connector_mqtt:reconnect:141 Reconnecting
```

gotthardp, 2019-05-07:
The VerneMQ is refusing the connection. There may be additional information in the VerneMQ logs. Is there something?

nicoinn, 2019-05-07:
The VerneMQ logs are clean.

More background: I add the let's encrypt certificate by doing a `cat ... >> /etc/ssl/certs/ca-certificates.crt`

If I don't do it, I get pretty much the same errors BUT the count is incrementing and the messages don't reach MQTT.

After adding the certificate, everything seems to work fine for my application, except for the above message, this time with a "count" always 0. 

But anyway, I can live with it - annoying, but not critical :)

gotthardp, 2019-05-09:
I see. That feels like you get connected and then (after 30 sec) disconnected. Is there a way to try increasing the required keepalive interval on VerneMQ? (It cannot be configured on the server-side unfortunatelly.)