gotthardp/lorawan-server #633: MQTT Connector does not reconnect after failure

hallard, 2019-05-09:
I noticed some time ago that as soon a MQTT connector is broken (example stopping mosquitto server for a certain amount of time) then the connector still never reconnect

[image]

Then I need to edit the connector
[image]

And then delete the network failure item and click save change 
[image]

Is there an option to auto-reconnect or at least send a email when this happens? Checking Send alerts on user email does the job?

gotthardp, 2019-05-11:
This is a known issue. Some errors are permanent, but shouldn't.

Enemoino, 2021-05-18:
Hi, is this issue beeing solved somehow ? Is it erlang version related issue ?

altishchenko, 2021-05-18:
> Hi, is this issue beeing solved somehow ? Is it erlang version related issue ?

No it is not an Erlang related issue, this is the way network server treats connectors internally. This issue was not solved yet, it may be fixed in the future if time permits.