gotthardp/lorawan-server #634: Access was denied You don't have authorization to view this page.

endermaxximum, 2019-05-09:
I'm delete my admin account in the server and it say

Access to (my ip) was denied You don't have authorization to view this page.
HTTP ERROR 403

gotthardp, 2019-05-09:
You need to follow the "I lost my password" troubleshooting guide
https://github.com/gotthardp/lorawan-server/blob/master/doc/Troubleshooting.md#lost-admin-password

endermaxximum, 2019-05-10:
I'm follow this command `connect to the cluster via remote shell by erl -sname test -remsh lorawan@<hostname>` but it show Erlang/OTP 21 [erts-10.3.5] [source] [64-bit] [smp:1:1] [ds:1:1:10] [async-threads:1]

*** ERROR: Shell process terminated! (^G to start new job) ***

User switch command
 --> mnesia:delete_table(user)
Unknown command
 --> 
BREAK: (a)bort (c)ontinue (p)roc info (i)nfo (l)oaded
       (v)ersion (k)ill (D)b-tables (d)istribution

Apaisal, 2019-05-10:
It seem server doesn't has test node name. If I change it with lorawan node
name then erl shell will be prompt.

gotthardp, 2019-05-10:
"test" is the name of the node being started (the console). I think the problem is the <hostname>. You shall use a hostname, not an IP address. Type `uname -n` and use that name.

endermaxximum, 2019-05-10:
Erlang/OTP 21 [erts-10.3.5] [source] [64-bit] [smp:1:1] [ds:1:1:10] [async-threads:1]

Eshell V10.3.5  (abort with ^G)
(lorawan@sl-thing-network)1> mnesia:delete_table(user).
{aborted,{node_not_running,'lorawan@sl-thing-network'}}
(lorawan@sl-thing-network)2> mnesia:start().           
ok
(lorawan@sl-thing-network)3> mnesia:delete_table(user).
{aborted,{no_exists,user}}
(lorawan@sl-thing-network)4> 

It said no exists,user

gotthardp, 2019-05-10:
The server must be running when you do that. Then you have stop the server and start it again to recreate the user database. (I assume you have some of the newer server versions. In older versions the table was called `users`. You may want deleting that too.)

endermaxximum, 2019-05-11:
When the server was running I cannot access it shows:
 sudo erl -sname lorawan -remsh lorawan@sl-thing-network
Protocol 'inet_tcp': the name lorawan@sl-thing-network seems to be in use by another Erlang node

gotthardp, 2019-05-12:
Please note that when running with `sudo` (which is not necessary) you run it with root credentials. Is the `.erlang.cookie` of root identical with the one used by the server?

gotthardp, 2019-05-12:
Try using `erl -sname test -remsh lorawan@sl-thing-network`. You changed `test` to `lorawan`, which is wrong.

endermaxximum, 2019-05-12:
I'm trying to use  erl -sname test -remsh lorawan@sl-thing-network
but it show like this
Erlang/OTP 21 [erts-10.3.5] [source] [64-bit] [smp:1:1] [ds:1:1:10] [async-threads:1]

*** ERROR: Shell process terminated! (^G to start new job) ***

gotthardp, 2019-05-12:
Did you verify the erlang cookies match?

endermaxximum, 2019-05-12:
/var/lib/lorawan-server/.erlang.cookie  **have but**  $HOME/.erlang.cookie **don't have**

gotthardp, 2019-05-12:
Copy it :)

endermaxximum, 2019-05-12:
erl -sname test -remsh lorawan@sl-thing-network
Erlang/OTP 21 [erts-10.3.5] [source] [64-bit] [smp:1:1] [ds:1:1:10] [async-threads:1]

*** ERROR: Shell process terminated! (^G to start new job) ***

I'm facing the same problem

gotthardp, 2019-05-12:
Anything suspicious in the server log?

endermaxximum, 2019-05-12:
crash.log    crash.log.1  crash.log.3  debug.log    debug.log.1  debug.log.3  error.log    error.log.1  error.log.3
crash.log.0  crash.log.2  crash.log.4  debug.log.0  debug.log.2  debug.log.4  error.log.0  error.log.2  error.log.4

**from error log**
2019-05-12 11:02:53.861 [error] <0.3753.0> ** Connection attempt from disallowed node 'test@sl-thing-network' **
2019-05-12 11:07:13.932 [error] <0.3776.0> ** Connection attempt from disallowed node 'test@sl-thing-network' **
2019-05-12 11:07:37.554 [error] <0.3778.0> ** Connection attempt from disallowed node 'test@sl-thing-network' **
2019-05-12 17:19:05.884 [error] <0.5086.0> ** Connection attempt from disallowed node 'test@sl-thing-network' **
2019-05-12 17:19:41.689 [error] <0.5088.0> ** Connection attempt from disallowed node 'test@sl-thing-network' **
**from crash.log** 
2019-05-12 11:02:53 =ERROR REPORT====
** Connection attempt from disallowed node 'test@sl-thing-network' **
2019-05-12 11:07:13 =ERROR REPORT====
** Connection attempt from disallowed node 'test@sl-thing-network' **
2019-05-12 11:07:37 =ERROR REPORT====
** Connection attempt from disallowed node 'test@sl-thing-network' **
2019-05-12 17:19:05 =ERROR REPORT====
** Connection attempt from disallowed node 'test@sl-thing-network' **
2019-05-12 17:19:41 =ERROR REPORT====
** Connection attempt from disallowed node 'test@sl-thing-network' **

gotthardp, 2019-05-12:
It's the `.erlang.cookie` problem. The same file shall be store in a **home directory** of the **user running** the lorawan-server and the user running the `remsh` command. For some reason it is not in your case.

endermaxximum, 2019-05-12:
So if I need to clean remove and reinstall the server again which file or folders that I have to delete?

gotthardp, 2019-05-12:
You can try stopping server and just removing the `user*` files from the mnesia directory in `/var/lib/lorawan-server/`

endermaxximum, 2019-05-12:
I have already done it but it still not work

gotthardp, 2019-05-12:
Then delete the entire content of `/var/lib/lorawan-server/`.

endermaxximum, 2019-05-12:
you mean delete and restart the server again?

gotthardp, 2019-05-12:
stop, delete and start.

endermaxximum, 2019-05-12:
sudo systemctl status lorawan-server.service
● lorawan-server.service - LoRaWAN Server
   Loaded: loaded (/lib/systemd/system/lorawan-server.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2019-05-12 21:31:50 UTC; 5s ago
  Process: 2401 ExecStart=/usr/lib/lorawan-server/bin/lorawan-server (code=exited, status=2)
 Main PID: 2401 (code=exited, status=2)

May 12 21:31:50 sl-thing-network systemd[1]: Started LoRaWAN Server.
May 12 21:31:50 sl-thing-network lorawan-server[2401]: /usr/lib/lorawan-server/bin/lorawan-server: 26: cd: can't cd to /
May 12 21:31:50 sl-thing-network systemd[1]: lorawan-server.service: Main process exited, code=exited, status=2/INVALIDA
May 12 21:31:50 sl-thing-network systemd[1]: lorawan-server.service: Failed with result 'exit-code'.

It's fail

gotthardp, 2019-05-12:
Did you delete `/var/lib/lorawan-server` or `/usr/lib/lorawan-server`?

endermaxximum, 2019-05-12:
l just delete only /var/lib/lorawan-server

gotthardp, 2019-05-12:
The content of, not the entire directory, right?

endermaxximum, 2019-05-12:
I delete the lorawan-server directory in **/var/lib/**

did you mean this?

gotthardp, 2019-05-12:
Then delete **the entire content of** /var/lib/lorawan-server/, not the directory itself.

endermaxximum, 2019-05-12:
sorry but which the directory that you mean?

gotthardp, 2019-05-12:
You need to start the server with an empty `/var/lib/lorawan-server` directory.

endermaxximum, 2019-05-12:
● lorawan-server.service - LoRaWAN Server
   Loaded: loaded (/lib/systemd/system/lorawan-server.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sun 2019-05-12 21:57:32 UTC; 4s ago
  Process: 2660 ExecStart=/usr/lib/lorawan-server/bin/lorawan-server (code=exited, status=1/FAILURE)
 Main PID: 2660 (code=exited, status=1/FAILURE)

closed 2019-05-31 by gotthardp