gotthardp/lorawan-server #636: How many devices ?

Ilheu, 2019-05-10:
Hi, 
I would like to implement your lorawan-server in my project, which consists of sending several sensor telemetry to the Azure Iot Hub. Can your solution handle many devices (thousands) ? 
(I know that this is not a issue, but I have no permission to acess  lorawan-server mailing list that you mention).

gotthardp, 2019-05-11:
Hello. The server itself can easily handle thousands on devices. However, you may find the web-admin features insufficient for handling such amount of devices.

Ilheu, 2019-05-14:
Hello,
Thanks for the answer !
Can you clarify a question for me? How do I generate Device DevEUI for each
device, in other Network Servers, these keys are generated by them.

gotthardp, 2019-05-14:
DevEUI is (also in this server) auto-generated upon device Join.

Ilheu, 2019-05-14:
Oh ok, thank you for your feedback !
Can you explain me what is the best workflow ( or creation order) in order
to configure the server ? I already understood that I need to create a
group before the devices, but before that I should create a network, is
that right ? Do you have any order ? At least for an simple example with 10
devices in one network/group ...
Greetings !

gotthardp, 2019-05-15:
The Configuration Guide shows that order
https://github.com/gotthardp/lorawan-server/blob/master/doc/Configuration.md#run-time-parameters

Ilheu, 2019-05-15:
I think that you already noticed it, I get this error, is there any
solution to solve it ?

[image: image.png]

gotthardp, 2019-05-15:
I don§t see the screenshot

Ilheu, 2019-05-15:
Sorry ...

Ilheu, 2019-05-15:
[image]

gotthardp, 2019-05-15:
I guess it is because you didn't set the **Google API Key**
https://github.com/gotthardp/lorawan-server/blob/v0.6.7/doc/Server.md#configuration

Ilheu, 2019-05-16:
I already connected one gateway, but how can I generate devices' NwkSKey and AppSKey ?

gotthardp, 2019-05-16:
These are created by the device manufacturer.

Ilheu, 2019-05-16:
I can get the APPEUI and DEVEUI trough the device's Lora Module (in this case rn2483), but I think that the AppKey should be generated by the server ... 

"AppKey: The AppKey is the data encryption key used to "encode" the messages between the end nodes and the Application Server. The AppKey is typically a random generated number who is programmed into the end nodes and also communicated to the Application server (so that it can decode the messages). " said in https://www.centennialsoftwaresolutions.com/blog/deveui-appeui-joineui-and-appkey

gotthardp, 2019-05-16:
AppKey is a unique key that is generated by whoever flashes the firmware to the device. Server just stores and uses this value.

Ilheu, 2019-05-16:
But in The Things Network, it is generated by the server ... 
[image]

gotthardp, 2019-05-16:
It's a different implementation. :-) Some people buy an off-the-shelf device and cannot change the AppKey, so there definitely must be an option to set whatever AppKey is hardcoded in the device.

What exactly is your issue? You can use whatever is in the device, or you can generate your own number. It can be any number, so you're free to use e.g. https://www.browserling.com/tools/random-hex to generate a list of 32-digit numbers.

gotthardp, 2019-05-16:
People that have many devices are very often using the HTTP API to configure their devices, so they are generating the AppKey via their script, e.g. by `tr -dc 'A-F0-9' < /dev/urandom | head -c32`.

Ilheu, 2019-05-16:
We made our device, but if it can be a random generated one that's ok ... We are trying to join our device in the network server, but we are always receiving denied message. And some times I get http_errors , like this image : 
[image]

I think I configured everything like your documentation says, I configured an Commissied device like the image below : 

[image]

Do I need to create an Activated node too ?!

gotthardp, 2019-05-16:
Have a look on the Event List to see some details for that error. Those errors are generated by the web-admin when you try to access the resource that doesnt exist. Its not related to device to server communication.

gotthardp, 2019-05-16:
When you are using OTAA (your devices send Join), then the Activated node is auto-created. Having a Commisioned device like you have should be enough.

gotthardp, 2019-05-16:
You should see at least a "Join" frame from the device. When you don't see really anything, then the device didn't send anything, the GW didn't receive it or didn't send it to the server.

Ilheu, 2019-05-16:
Than you for your help ! I think it might be a problem in the gateway, we are receiveing the same error connecting to TTN !

Ilheu, 2019-05-17:
Sorry to be annoying, but can you help me to debug this ? I can connect with the same device and gateway to the TTN, but can't connect your network server  ... 
Is there any way to check if the gateway is receiving any package ?

gotthardp, 2019-05-17:
No problem :)
1) Does the server say "connected" in the timeline when you start the gateway? The gateway shall have a green "check".
2) The gateway has its own logs. Are there any errors/undelivered frames logged?

Ilheu, 2019-05-17:
1. 
[image]

2. Are you talking about the status tab or in the gateway interface ?
[image]

I using a Dragino, I don't know if it has any issue with your network server ...

gotthardp, 2019-05-17:
This looks good. You should really check the gateway and server logs to see if the frame was received and what happened afterwards.

Ilheu, 2019-05-17:
Finally found the logs ...

[image]

gotthardp, 2019-05-17:
Yeah, "RF packets received by concentrator: 0" Is there anywhere some higher number? If not, then the gateway is not receiving anything (and thus cannot send it to the server). Not sure why it works with TTN though.

Ilheu, 2019-05-17:
I'm receiving ...

[image]

gotthardp, 2019-05-17:
Alright, good. Then there should be something in the server logs. Could you check that, please?

Ilheu, 2019-05-17:
Finally we did it ! Thank You so much ! 
[image]

gotthardp, 2019-05-17:
Cool! I am glad it works.

closed 2019-05-17 by gotthardp