gotthardp/lorawan-server #637: Device do not show up as unknown devices once Deleted

Tamburasca, 2019-05-17:
We see the strange behavior that devices do not show up in the dashboard anymore once they were deleted from the activated nodes. Is this desired?

Cheerio, tamburasca

gotthardp, 2019-05-17:
Where would you expect the node to show after being deleted?

Tamburasca, 2019-05-17:
[image]

here as red bar - unknown device

gotthardp, 2019-05-17:
Yeah, that would be logical, but it doesn't work that way. The frames are deleted and don't show as "unknown".

Tamburasca, 2019-05-17:
And this applies also in case the gateway still sees the devices

gotthardp, 2019-05-17:
When a new frame from the deleted node arrives, it should be displayed as "unknown" (if logging of unknown frames is enabled)

Tamburasca, 2019-05-17:
good to know, I haven't gotten that far yet! Thanks.

Tamburasca, 2019-05-17:
I trawled through your documentation, however, did not find how to enable unknown frames. Sorry for being persistent.

gotthardp, 2019-05-17:
Under Area, there is "Log Ignored"