gotthardp/lorawan-server #638: Can't join LAIRD Lora module to  Lorawan Server

Matavulj, 2019-05-19:
Dear Gotthard,
I tried to connect  LAIRD Development kit (DVK-RM186) with LAIR gateway RG186  to Loriot Lorawan server and everything was OK. I can see sent text data. After that I tried the same with your Lorawan server  but without succeed.   Gateway connects to Lorawan server but module can't join to network. I can see some crash in Logs. I installed Windows version v0.6.7 and Erlang 19. 
Please help, I don't know why server crashed. I send you some pictures (adjustments) and logs in attachment. See time 19.05.2019 at 10:15
Matavulj
[crash.log](https://github.com/gotthardp/lorawan-server/files/3194905/crash.log)
[debug.log](https://github.com/gotthardp/lorawan-server/files/3194906/debug.log)
[image]
[image]
[image]
[error.log](https://github.com/gotthardp/lorawan-server/files/3194907/error.log)
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

gotthardp, 2019-05-19:
Please send me the crash logs.

Matavulj, 2019-05-19:
Petr,
Thanks for quick answer. You have crash.log in the beginning of  this issue, billow my signature and the first picture.

Matavulj, 2019-05-19:
I send you error.log also in this attachment. You have also **debug.log**    with **crash.log** in the beginning of this issue.
[error.log](https://github.com/gotthardp/lorawan-server/files/3195472/error.log)

gotthardp, 2019-05-19:
This seems to be Erlang issue. You have Erlang/OTP 19.0, right? Please try upgrading to 19.1 (or later).

Matavulj, 2019-05-20:
Hi Petr,
 I installed Erlang 22 and now everything is OK. I can see data at Frames form.
Thanks a lot.

gotthardp, 2019-05-21:
Cool :)

closed 2019-05-21 by gotthardp