gotthardp/lorawan-server #639: ABP messages

Ilheu, 2019-05-20:
After the device joined via OTAA, the server does not receive any message from the node. After OTAA the devices goes to sleep and then wakes up. Since the device has already joined via OTAA it uses ABP to join and send a message from the sensor. 

The gateway does not receive all sended messages, just some. I'm trying to send message each minute.

[image]

I'm always receiving this message, but it is empty : 

[image]

gotthardp, 2019-05-20:
What exactly is the issue, please? "the server does not receive any message from the node" or "gateway does not receive all sended messages, just some"?

Ilheu, 2019-05-20:
The server is not receiving any ABP message, but receives OTAA joins ...

gotthardp, 2019-05-20:
That may be a Network configuration issue.

Ilheu, 2019-05-20:
Is it possible to use a Initial Duty Cycle which is not in your options , for example, 0.33 % ?

gotthardp, 2019-05-20:
No, it has a to be a 2^x fraction. This is a protocol limitation, not a server constraints.

Ilheu, 2019-05-21:
We are receiving already ABP messages, but we get the error "not_semtech_mote" with the message. In the device profile I only have semtech-mote option.  We are using Microchip's RN2483 module. Can we fix this error, or it is not important ?

Btw, how can I transfer my lorawan-server to another computer, I'm using a Windows pc, and I want to transfer the server configuration to linux, is it possible ?

gotthardp, 2019-05-21:
For the "not semtech mote" error, please see https://github.com/gotthardp/lorawan-server/blob/master/doc/Troubleshooting.md#not_semtech_mote

To transfer the config, please use the Export/Import scripts https://github.com/gotthardp/lorawan-server/blob/master/doc/Administration.md#backup-and-restore

Ilheu, 2019-05-21:
I did bin/dbexport but the created file is empty ...

gotthardp, 2019-05-21:
Any error in the log?

Ilheu, 2019-05-21:
I'm using windows 10, the cmd window dissapears ...

gotthardp, 2019-05-21:
I meant server log.
You can also open the window using `cmd` then `cd` to the right directory and run the script.

Ilheu, 2019-05-21:
I already had success in the import ! Thank you ! But I still get the "not semtech mote" error.  Do I need to add an aplication for my semtech-mote ?

gotthardp, 2019-05-22:
What application is running on your RN2483? The "semtech-mote" is a demo app from https://github.com/Lora-net/LoRaMac-node/blob/master/src/apps/LoRaMac/classA/LoRaMote/main.c#L207

If you are running somethng else you need to define a custom app. (It is safe to ignore the error but the server will not do anything useful.)

Ilheu, 2019-05-22:
What you mean with "but the server will not do anything useful." ?!

gotthardp, 2019-05-22:
You want to handle the application payload somehow, right? To do that you need to define an application.

Ilheu, 2019-05-22:
Yes I need to handle application payload ! I've been reading your documentation, and I have to implement a microchip-mote application for Microchip LoRa(TM) Technology Mote, which is available at https://github.com/gotthardp/lorawan-server-demoapp. But I don't know/understand what I have to do .... Is there any tutorial ?, I just want to read the messages' data in IotHub ...

gotthardp, 2019-05-22:
You can also send the payload to some external server via MQTT etc.

Ilheu, 2019-05-22:
I just want to save the messages in azure iotHub ...

gotthardp, 2019-05-22:
Then you don't need to develop anything. Just follow the https://github.com/gotthardp/lorawan-server/blob/master/doc/Integration.md#microsoft-azure-iot-hub

Ilheu, 2019-05-22:
My connector has status: failed .... 

My node is : 

[image]

My connector general:

[image]

My connector Auth: 

[image]

gotthardp, 2019-05-22:
It's the `Failed` flags. You can delete them by hand as it may be caused due to some previous activity. If they appear again you should check the logs and fix the problem before deleting the flag again.

Ilheu, 2019-05-22:
I deleted the flags, but I'm not receiving anything in azure iothub ... 

 - is devaddr in the "devices/{devaddr}/messages/events/", the node DevAddr ?

gotthardp, 2019-05-22:
Yes.

Ilheu, 2019-05-22:
Do you have any idea why I am not receiving the messages ?

Ilheu, 2019-05-22:
Do I need a handler ?

gotthardp, 2019-05-22:
Of course :)

Ilheu, 2019-05-22:
oh ok ... it is not mentioned in https://github.com/gotthardp/lorawan-server/blob/master/doc/Integration.md#microsoft-azure-iot-hub ...

Ilheu, 2019-05-22:
I just want to send a JSON to Azure Iot Hub, what language are you using in Parse Uplink, Parse Event and Build Downlink fields ? Do I need to use them if yes, is there any documentation how to do it ?

[image]

Ilheu, 2019-05-22:
Can I test the handler ? Or just when I receive messages ?

Ilheu, 2019-05-22:
The documentation of Azure IotHub integration is updated ?

gotthardp, 2019-05-22:
The documentation should be up to date. The handler funs should be written in Erlang, as documented here: https://github.com/gotthardp/lorawan-server/blob/master/doc/Handlers.md#parse-uplink

There is not direct debug functionality, but you can use an online tool as explained in https://github.com/gotthardp/lorawan-server/issues/625

Ilheu, 2019-05-22:
I read that I don't need a parse uplink ... So I don't get why it is not connecting with my iotHub, can you check the pictures above ?

gotthardp, 2019-05-22:
I cannot see anything suspicious there. Have a look into the server and iotHub logs.

Ilheu, 2019-05-22:
When you say server logs, you mean the events tab ?

gotthardp, 2019-05-22:
I meant the text file with detailed logs: https://github.com/gotthardp/lorawan-server/blob/master/doc/Troubleshooting.md#troubleshooting-instructions

The iotHub might also have some debugging log showing what went wrong.

Ilheu, 2019-05-23:
I don't see anything unusual in the server logs :

<<"87945D">>
2019-05-23 10:41:40.244 [debug] <0.2922.2>@lorawan_handler:join:114 Join-Accept in RX1: {rxq,868.3,<<"SF9BW125">>,<<"4/5">>,{{2019,5,23},{9,41,38.916699}},undefined,3816871322,-58,7.8}
2019-05-23 10:42:10.185 [info] <0.2954.2> device 0004A30B00E8AB91 {join,<<"ABC13456">>}
2019-05-23 10:42:10.186 [warning] <0.2954.2> node ABC13456 {repeated_reset,7}
2019-05-23 10:42:10.189 [debug] <0.2954.2>@lorawan_mac:encode_accept:441 Join-Accept <<"ABC13456">>, netid <<0,0,0>>, cflist [], rx1droff 0, rx2dr 3, appkey <<"C82DB0DC7CA167475F8E382B33D668A9">>, appnce <<"884A6D">>
2019-05-23 10:42:10.189 [debug] <0.2954.2>@lorawan_handler:join:114 Join-Accept in RX1: {rxq,868.3,<<"SF9BW125">>,<<"4/5">>,{{2019,5,23},{9,42,8.948648}},undefined,3846903271,-59,7.8}
2019-05-23 10:49:18.343 [debug] <0.3296.2>@lorawan_mac_commands:store_actual_adr:147 DataRate ABC13456 switched to dr 3
2019-05-23 10:49:18.344 [debug] <0.3296.2>@lorawan_mac_commands:build_fopts:77 ABC13456 <- [dev_status_req]
2019-05-23 10:49:18.345 [error] <0.3296.2> node ABC13456 {not_semtech_mote,{1,<<"993">>}}

Have you ever used Azure Iot Hub ?
I'm not receiving anything ....

[image]

Ilheu, 2019-05-23:
I solved it, I had to change the application in the device profile to my handler ...

gotthardp, 2019-05-23:
Cool. Shall I close the issue then?

closed 2019-05-23 by Ilheu