gotthardp/lorawan-server #640: message not publish to out/{deveui} topic

elisascheer, 2019-05-22:
Hi,

I declare an Application as follow : 
[image]

and I declared the associated profile as follow : 

[image]

I declared my device which uses the Temperature profile created and I created my mqtt connector : 

[image]

But when I subscribe to the topic
`mosquitto_sub -h 127.0.0.1 -p 1883 -t 'out/343531314F377213' -u myuser -P mypass`

nothing published and frames are received : 

[image]

elisascheer, 2019-05-22:
In the debug.log there is this line :
[image]

gotthardp, 2019-05-22:
Which server version are you using. There used to be a bug in the template handling.

elisascheer, 2019-05-22:
I am running 0.6.6

gotthardp, 2019-05-22:
I see now where the problem is: you have to pass the `deveui` through your parsing function. Instead of `#{fields...}` do `Fields#{fields...}` (prepend Fields) in all parts of your parsing fun.

elisascheer, 2019-05-22:
Thanks it worked! :)

closed 2019-05-22 by elisascheer