gotthardp/lorawan-server #641: uplink_missand and downlink_missed, and can not confirm ack

vinsanitycarter, 2019-05-27:
I am having uplink_missed and downlink_missed error.
[image]

First of all,  let me know the test enviroment.

[End device setting]
- hardware : RAK811
- software : RAK811 AT Command
- band : US902-915
- actiation mode : abp
- data rate : DR1
- Payload size : 14
- channel : 0 to 7
- rx2 channel : channel 8, 923.3Mhz
- rx1 delay : 1000ms
- radio power : 20dBm
[image]
[image]

[gateway setting]
- hardware : raspberry PI 3 + RisingHF rhf0m301
- sofware : Official RisingHF Packet Forwarder

Here is global_conf.json

       "SX1301_conf":{
        "lorawan_public": true,
        "clksrc": 1, /* radio_1 provides clock to concentrator */
        "antenna_gain": 0, /* antenna gain, in dBi */

        "radio_0": {
            "enable": true,
            "type": "SX1257",
            "freq": 902700000,
            "rssi_offset": -166.0,
            "tx_enable": true,
            "tx_freq_min": 902000000,
            "tx_freq_max": 928000000
        },

        "radio_1": {
            "enable": true,
            "type": "SX1257",
            "freq": 903400000,
            "rssi_offset": -166.0,
            "tx_enable": false
        },

        "chan_multiSF_0": {
            /* Lora MAC channel, 125kHz, all SF, 902.3 MHz */
            "enable": true,
            "radio": 0,
            "if": -400000
        },
        "chan_multiSF_1": {
            /* Lora MAC channel, 125kHz, all SF, 902.5 MHz */
            "enable": true,
            "radio": 0,
            "if": -200000
        },
        "chan_multiSF_2": {
            /* Lora MAC channel, 125kHz, all SF, 902.7 MHz */
            "enable": true,
            "radio": 0,
            "if": 0
        },
        "chan_multiSF_3": {
            /* Lora MAC channel, 125kHz, all SF, 902.9 MHz */
            "enable": true,
            "radio": 0,
[... 246 lines cut ...]
[image]
[image]
At "at+recv=1,0,0" message, if first parameter is 1, confirmed message transfer was successful and ack signal is received from server.

Downlink was also done normally.
[image]
[image]

At at+recv=0,2,-40,26,5,0908070605
0: Indication that data was received
2: port
-40 : rssi
26 : snr
5 : data length
0908070605 : data

this picture is gateway traffic
[image]

this gateway hardware is raspberry pi3 + risingHF rhfm031, 
and software is TTN Packet Forwarder on a raspberry pi with and IMST ic880a board.

There seems to be a problem with the gateway or the lorawan-server. 
First, I will check the lorawan-server.
Is there anything wrong with the server settings?

gotthardp, 2019-05-27:
What a great and throughout report! I love it.

From the channel setting it seems you want to downlinks to use a single channel (64). To achieve that you should use the "US 902-928MHz **Private Hybrid**" region. Try that, please.

vinsanitycarter, 2019-05-28:
> What a great and throughout report! I love it.
> 
> From the channel setting it seems you want to downlinks to use a single channel (64). To achieve that you should use the "US 902-928MHz **(Private Hybrid)**" region. Try that, please.

Thank you for answer.
I didn't think to use a single channel(64).
Just i set it to [0-7, 64] because uplink uses channel 0-7 channel.
But i tried changing the region as you told me. And test it.
The results are nothing new.
I also tested the changing the region to [0-71].
There was nothing that changed either.
It seems that setting the server's region is not a problem.
Is there anything else that could be a problem?

vinsanitycarter, 2019-05-28:
> What a great and throughout report! I love it.
> 
> From the channel setting it seems you want to downlinks to use a single channel (64). To achieve that you should use the "US 902-928MHz **(Private Hybrid)**" region. Try that, please.
Additional testing has confirmed that this is not a gateway issue. There is a greater chance of a problem on the server side.

gotthardp, 2019-05-28:
I have no idea what could be wrong, unfortunately. These types of errors are hard to debug remotely.