gotthardp/lorawan-server #642: downlink - RX2 window - tx_freq error - class C

gitposter, 2019-06-18:
[image]

Hello,
Class A works fine (up and downlink) but I have issue sending downlinks in Class C mode (RX2 window). I'm using E433 region default settings.
[image]
[image]

My setup is as follows: RPi (server and gw) and stm32 (node), all are equiped with sx1276 modules.

could any one help please ?

gotthardp, 2019-06-18:
This is a gateway-generated error. Have a look in the log of your gateway. There should be some more details about the rootcase.

From what you shared you have RX2 Freq 434.0, whereas by default it should be 434.665, see https://github.com/gotthardp/lorawan-server/blob/master/doc/Regions.md

gitposter, 2019-06-18:
Thank you for replying,

nothing new, log is:
`2019-06-18 11:01:33.886 [error] <0.324.0> node 26011A71 <<"tx_freq">>
2019-06-18 11:04:36.897 [error] <0.324.0> node 26011A71 <<"tx_freq">>
`
is there any way to force higher verbose level ?

gotthardp, 2019-06-18:
this is a **server** log. I meant a **gateway** log

gitposter, 2019-06-18:
sorry for that, the error is in the used frequency
`JSON down: {"txpk":{"powe":14,"imme":false,"tmst":4047297573,"codr":"4/5","datr":"SF12BW125","freq":867.5,"modu":"LORA","rfch":0,"ipol":true,"size":23,"data":"YHEaASaLGgADSgcAAAUAGlNCBsveZno="}}
ERROR: Packet REJECTED, unsupported frequency - 867500000 (min:433050000,max:434790000)
INFO: [down] PULL_ACK received in 0 ms`
my configuration file is:
[image]

gotthardp, 2019-06-18:
Somewhere in your configuration there is "867.5". You should find it and replace by a correct number.
Or, your device used 867.5, whereas the gateway is configured on 434. In any case, its a configuration issue.

gitposter, 2019-06-18:
I think the configuration is good, the odd frequency (867) is depicted from the packet received from server:
`JSON down: {"txpk":{"powe":14,"imme":false,"tmst":1882915003,"codr":"4/5","datr":"SF12BW125","freq":867.5,"modu":"LORA","rfch":0,"ipol":true,"size":23,"data":"YHEaASaLcgMDSgcAAAUAGlNCBgMvfKk="}}
ERROR: Packet REJECTED, unsupported frequency - 867500000 (min:433050000,max:434790000)
`
Could you guide me where should I dig in server's code please ?

gotthardp, 2019-06-18:
Someone must have used this frequency: either the sending device is wrong, or the server configuration. There are no hardcoded frequencies in the code. Do you have the equivalent `rxpk` field? Please check the `freq` there.