gotthardp/lorawan-server #643: Multiple send

Ilheu, 2019-06-24:
Hi, 
Suddently, I start receiving in my IotHub multiple messages from each end device, when a single node sends a message. But in the messages sent, the devaddr is from the real one. Some of the nodes are not working anymore, so they can't be sending data ...

More simply, I receive a message from device 1, and it sends 10 identical messages to the IotHub for each existing device ...

gotthardp, 2019-06-24:
It's likely due to a connector misconfiguration.

Ilheu, 2019-06-24:
I have 10 connectors, with the same handler. 

[image]

[image]

[image]

I think it is alright ...

gotthardp, 2019-06-24:
Yes, that is as expected. With 10 connectors you get 10 messages. What is the issue?

Ilheu, 2019-06-24:
I receive a message from ONE device, and send TEN messages with the same content and devadrr, but from each device ... So, the device 1 send me a message, and for example, the device 7 sends it too the same message ...
Like this ...
[image]

gotthardp, 2019-06-24:
Isn't the DevAddr fixed for each connector? I see connector1 is linked with `ABC13456`. You should use the template `{DevAddr}` instead.

Ilheu, 2019-06-24:
You right !! Thank you !!!

closed 2019-06-24 by gotthardp