gotthardp/lorawan-server #644: OTAA activation : issue with join-accept

quentinchanu, 2019-06-25:
Hello and thanks for your work,

I have trouble activation my node (lopy4) with OTAA on your server, (but it's working with TTN)

The warning i get in the dashboard is 'repeated reset'. 

Indeed at the first uplink of my node the server activated it correctly and sended the downlink join-accept but it looks like my node doesn't understand it because he keep sending join-request.
So now i am stuck in a loop with my node sending join-request and the server showing warnings repeated-reset.

Important fact : my gateway forward the downlink packet join-accept to my node ! Here is the terminal of my gateway and the debug.log file 

[image]

[image]

I hope you can help me and I suppose I have not done mistake on my settings parameters but feel free to ask and i will answer you ! I was thinking of an hypothesis maybe the node cannot decode the downlink or maybe it's not encode properly ? Where can be my mistake ?

gotthardp, 2019-06-25:
I suspect the issue is related to the downlink frequency. Did you correctly set the Region and the RX2 Freq?

quentinchanu, 2019-06-25:
Yes i think that my settings are correct : 

[image]

gotthardp, 2019-06-25:
`868,1` is not inline with the recommended defaults. Are you aware of this?
https://github.com/gotthardp/lorawan-server/blob/master/doc/Regions.md

quentinchanu, 2019-06-25:
I just changed the Rx2 frequency and Rx2 DR to respect the recommended defaults and it's still not working. Is the join-accept supposed to always go through RX2 and never through RX1 ?
Where can the problem come from now ?
On the side of my server the node is activated but on the side of the node he is still sending join-request because it doesnt receive any join-accept.

gotthardp, 2019-06-25:
It's very likely caused by a misalignment of RF settings (DR, freq, timing) of the device, the gateway and the server. I can only recommend you to re-review the configuration again.

quentinchanu, 2019-06-26:
Yes you were right I didn't know that the Rx1 and Rx2 for join accept messages was not the same delay as classic downlink ! I fixed it and now its working ! I just have one more question and then you can close the issue : where can we set the CFlist to manage which channel to send, i found it was in the release 0.5.0 but i don't find where ? (tell me if you prefer me to open a new issue for this question)
Thanks again !

gotthardp, 2019-06-26:
It's in the network configuration, last tab

quentinchanu, 2019-06-26:
Last question : in the hypothesis I would like to use a single channel gateway can it be set to only one channel ? (I know that the specification is saying that you have to use at least the 3 first channels : 0-2)

gotthardp, 2019-06-26:
There were some attempts to do that. You'd need to set the device (profile) to either RX1 or RX2 window only. The channel list is rather a device config; the server has not much to do with that.

closed 2019-06-26 by quentinchanu