gotthardp/lorawan-server #646: uplink will automatically follow a downlink

houhoututu, 2019-06-28:
Hi,I have a problem, the node is Class C mode, and each uplink will automatically follow a downlink . How to get rid of this downlink,it will affect the capacity. Below is my server configuration and log.

[image]
[image]
configuration
[image]
[image]
[image]

gateway log

`pi@raspberrypi:~/lora/packet_forwarder/lora_pkt_fwd $ ./lora_pkt_fwd 
*** Beacon Packet Forwarder for Lora Gateway ***
Version: 4.0.1
*** Lora concentrator HAL library version info ***
Version: 5.0.1;
***
INFO: Little endian host
INFO: found global configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
INFO: lorawan_public 1, clksrc 1
INFO: no configuration for LBT
INFO: antenna_gain 0 dBi
INFO: Configuring TX LUT with 16 indexes
INFO: radio 0 enabled (type SX1255), center frequency 470600000, RSSI offset -166.000000, tx enabled 1, tx_notch_freq 0
INFO: radio 1 enabled (type SX1255), center frequency 471400000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
INFO: Lora multi-SF channel 0>  radio 0, IF -300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 1>  radio 0, IF -100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 2>  radio 0, IF 100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF 300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 4>  radio 1, IF -300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 5>  radio 1, IF -100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 6>  radio 1, IF 100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 7>  radio 1, IF 300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora standard channel 8 disabled
INFO: FSK channel 8 disabled
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to B827EBFFFE13F407
INFO: server hostname or IP address is configured to "192.168.8.30"
INFO: upstream port is configured to "1680"
INFO: downstream port is configured to "1680"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: found local configuration file local_conf.json, parsing it
INFO: redefined parameters will overwrite global parameters
INFO: local_conf.json does not contain a JSON object named SX1301_conf
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to B827EBFFFE13F407
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: [main] concentrator started, packet can now be received

INFO: Disabling GPS mode for concentrator's counter...
INFO: host/sx1301 time offset=(1561687211s:630452µs) - drift=-1551900812µs
INFO: Enabling GPS mode for concentrator's counter.

INFO: [down] PULL_ACK received in 1 ms
INFO: [down] PULL_ACK received in 1 ms

INFO: Received pkt from mote: 66666693 (fcnt=38)

JSON up: {"rxpk":[{"tmst":15752092,"chan":1,"rfch":0,"freq":470.500000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":9.2,"rssi":-83,"size":22,"data":"QJNmZmaAJgADcomcbXHN1ghxnBvKig=="}]}
INFO: [up] PUSH_ACK received in 3 ms
INFO: [down] PULL_RESP received  - token[206:43] :)

JSON down: {"txpk":{"powe":14,"imme":false,"tmst":16752092,"codr":"4/5","datr":"SF12BW125","freq":500.5,"modu":"LORA","rfch":0,"ipol":true,"size":13,"data":"YJNmZmYBHAAGqSB97g=="}}
INFO: tx_start_delay=1495 (1495.500000) - (1497, bw_delay=1.500000, notch_delay=0.000000)
INFO: [down] PULL_ACK received in 2 ms

 2019-06-28 02:00:44 GMT 
 [UPSTREAM] 
 RF packets received by concentrator: 1
 CRC_OK: 100.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
 RF packets forwarded: 1 (22 bytes)
 PUSH_DATA datagrams sent: 1 (205 bytes)
 PUSH_DATA acknowledged: 100.00%
 [DOWNSTREAM] ###
 PULL_DATA sent: 3 (100.00% acknowledged)
 PULL_RESP(onse) datagrams received: 1 (173 bytes)
 RF packets sent to concentrator: 1 (13 bytes)
 TX errors: 0
 TX rejected (collision packet): 0.00% (req:1, rej:0)
 TX rejected (collision beacon): 0.00% (req:1, rej:0)
 TX rejected (too late): 0.00% (req:1, rej:0)
 TX rejected (too early): 0.00% (req:1, rej:0)
 BEACON queued: 0
 BEACON sent so far: 0
 BEACON rejected: 0
 [JIT] 
 SX1301 time (PPS): 2434781
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
 [GPS] 
 GPS sync is disabled
 END 

JSON up: {"stat":{"time":"2019-06-28 02:00:44 GMT","rxnb":1,"rxok":1,"rxfw":1,"ackr":100.0,"dwnb":1,"txnb":1}}
INFO: [up] PUSH_ACK received in 5 ms
INFO: [down] PULL_ACK received in 5 ms

INFO: Received pkt from mote: 66666693 (fcnt=39)

JSON up: {"rxpk":[{"tmst":33313100,"chan":0,"rfch":0,"freq":470.300000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":10.2,"rssi":-81,"size":22,"data":"QJNmZmaAJwADHJUnkc+lfFyOh8Qp0g=="}]}
INFO: [up] PUSH_ACK received in 1 ms
INFO: [down] PULL_RESP received  - token[216:148] :)

JSON down: {"txpk":{"powe":14,"imme":false,"tmst":34313100,"codr":"4/5","datr":"SF12BW125","freq":500.3,"modu":"LORA","rfch":0,"ipol":true,"size":13,"data":"YJNmZmYBHQAGGKTr3A=="}}
INFO: tx_start_delay=1495 (1495.500000) - (1497, bw_delay=1.500000, notch_delay=0.000000)
INFO: [down] PULL_ACK received in 12 ms

INFO: Received pkt from mote: 66666693 (fcnt=40)

JSON up: {"rxpk":[{"tmst":51880028,"chan":3,"rfch":0,"freq":470.900000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":7.2,"rssi":-77,"size":22,"data":"QJNmZmaAKAAD8+/TtMGOoXvF97QZxQ=="}]}
INFO: [up] PUSH_ACK received in 2 ms
INFO: [down] PULL_RESP received  - token[15:148] :)

JSON down: {"txpk":{"powe":14,"imme":false,"tmst":52880028,"codr":"4/5","datr":"SF12BW125","freq":500.9,"modu":"LORA","rfch":0,"ipol":true,"size":13,"data":"YJNmZmYBHgAG2lKFxg=="}}
INFO: tx_start_delay=1495 (1495.500000) - (1497, bw_delay=1.500000, notch_delay=0.000000)
INFO: [down] PULL_ACK received in 2 ms

INFO: Disabling GPS mode for concentrator's counter...
INFO: host/sx1301 time offset=(1561687211s:630458µs) - drift=6µs
INFO: Enabling GPS mode for concentrator's counter.
`

gotthardp, 2019-07-01:
The server must have a reason for sending the downlink. Please have a look in the server `debug.log` to see what is the server sending back to the device. After learning that we should be able to stop it.

houhoututu, 2019-07-02:
Check the debug.log  and find out

> 2019-07-02 13:56:58.328 [debug] <0.344.0>@lorawan_mac_commands:build_fopts:77 66666693 <- [dev_status_req]

The server sent a status query command.Measuring the gateway antenna, there is indeed a transmission, but the node did not receive this command, and naturally there is no return status.Check the log of the gateway to find that the downlink frequency is 30MHz higher than the uplink frequency.This does not match the receive window of RX1.

> JSON up: {"rxpk":[{"tmst":51880028,"chan":3,"rfch":0,"freq":470.900000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":7.2,"rssi":-77,"size":22,"data":"QJNmZmaAKAAD8+/TtMGOoXvF97QZxQ=="}]}
INFO: [up] PUSH_ACK received in 2 ms
INFO: [down] PULL_RESP received - token[15:148] :)
JSON down: {"txpk":{"powe":14,"imme":false,"tmst":52880028,"codr":"4/5","datr":"SF12BW125","freq":500.9,"modu":"LORA","rfch":0,"ipol":true,"size":13,"data":"YJNmZmYBHgAG2lKFxg=="}}
INFO: tx_start_delay=1495 (1495.500000) - (1497, bw_delay=1.500000, notch_delay=0.000000)
INFO: [down] PULL_ACK received in 2 ms

Then I tried to modify the profiles, TXwindow changed to RX2, which is fixed 505.3mzh.Now,the node received the dev_status_req and then replied.There is no downlink for the second uplink.

[image]

Question, why the server's sending frequency in the RX1 window has increased by 30MHz

gotthardp, 2019-07-02:
The Freq assignment depends on the Region. Which `Region` did you set?

houhoututu, 2019-07-03:
Hi,gotthardp. The region is CN470,I ignored some content, the downlink frequency is 500.3-509.7MHz.I mistakenly believe that the downlink frequency and the uplink frequency are the same in the RX1 window.After modifying the node settings, the command was successfully received.
Sincerely thank you for your reply.The problem can be closed.

closed 2019-07-03 by houhoututu