gotthardp/lorawan-server #647: 'parse failed' warning

quentinchanu, 2019-06-28:
Hi i would like the parse the 'data' values that i received on my MQTT topic to receive something like {« date » :….., « temp » :……….., « hum » :….. […] « batt » :………} instead of {« data » :……..}

Basically I receive 2 kinds of frames : (TimeSync request and one with my measures)
TimeSync looks like : {"data":"01FA9CA8240001190402","devaddr":"019CB2BA"}
Measure looks like : {"data":"0CB05CDB26CD09818E9101F5047500905EDB26E20980899101E804BD0063","devaddr":"019CB2BA"}

And here is how i would like to parse both of them :

[image]

[image]

Parsing code (Erlang)

[image]

Here is the code in the parse uplink section :

fun (Fields, <<16#0C:16, Date:64, Temp:32, Hum:16, Pressure:48, Lum:32, Air:32, Date2:64, Temp2:32, Hum2:16, Pressure2:48, Lum2:32, Air2:32,Batt:16>>) ->
  Fields#{date => Date, temp => Temp, hum => Hum, pressure => Pressure, lum => Lum, air => Air, date2 => Date2, temp2 => Temp2, hum2 => Hum2, pressure2 => Pressure2, lum2 => Lum2, air2 => Air2 , batt => Batt};
     (Fields, <<16#01:16, Idrequest:64, Version:48, Appid:32>>) ->
Fields#{idrequest => Idrequest, version => Version, appid => Appid}
end.

I don’t understand what have I done wrong because I get for every uplink the warning ‘parse_failed’ in the dashboard and when i do the following command in my terminal: 
mosquitto_sub -h 127.0.0.1 -p 1883 -t 'out/#' -u 'user' -P 'pass'
I always receive the following JSON {« data » :…………………….., « devaddr » :………} and the data is not parsed ! Is is possible to parse the data als I want ? Is my code wrong ?

Thanks a lot for your help and your work !

gotthardp, 2019-07-01:
Hello. It seems the size indicators are wrong. In a hexa-string (like in your examples) two hex-digits are one byte, i.e. 8 bits. So, `01FA9CA824` can be matched with `<<16#0C:8, Date:32>>`, because it is 5 bytes, i.e. 40 bits (8+32). In your patterns all the lengths are doubled. Why?

quentinchanu, 2019-07-04:
Thanks a lot it's working now !

closed 2019-07-04 by quentinchanu