gotthardp/lorawan-server #649: Websockets per device

Kralg, 2019-07-02:
Hello,

We are trying/analyzing how to connect this server to another over the websockets for UI. The thing is to have websockets clients per deveui/devaddr which will connect to the other server. I was reading Integration and this could probably be similar to the Thingsboard but not MQTT. Is there already a pattern on how to implement this?

gotthardp, 2019-07-02:
Hello. Outbound WebSockets (clients to a remote server) are not supported.

Kralg, 2019-07-02:
I see, can you write how complex this is to be implement as server enhancement? It would be good to roughly estimate how feasible is this to be supported and possible time when it could be used.

gotthardp, 2019-07-02:
Having a WebSocket is quite easy to implement even for a newbie. The per-device WebSockets are a bit more complex. I'd say few days, depending on Erlang proficiency.

Kralg, 2019-07-03:
I believe this is related to lorawan_connector_ws.erl module. Comparing it with lorawan_connector_mqtt.erl not so much of initial insights comes maybe only that supervisor seems to create the childs connectors (maybe). Thats still sort of challenge - as my skills of Erlang are more like a basics. If it is between the enhancements of the server can you write how long would it take for you to implement this on the server?

gotthardp, 2019-07-03:
I'd say 1-2 days, but in general I implement new features only when I see demand from multiple users.

closed 2024-05-09 by Kralg