gotthardp/lorawan-server #651: join ack

arigus, 2019-07-08:
Hi Peter,
I developing lorawan stack under freertos,  and encounter join ack missed at device side.
debug log as bellow:
2019-07-08 14:40:18.372 [info] <0.4188.0> device BE7A00000000031B {join,<<"AD7325D3">>}
2019-07-08 14:40:18.373 [warning] <0.4188.0> node AD7325D3 {repeated_reset,611}
2019-07-08 14:40:18.373 [debug] <0.4188.0>@lorawan_mac:encode_accept:441 Join-Accept <<"AD7325D3">>, netid <<18,52,86>>, cflist undefined, rx1droff 0, rx2dr 0, appkey <<"A3B79F51F639B2B43460707D2AA86D4A">>, appnce <<"A15624">>
2019-07-08 14:40:18.373 [debug] <0.4188.0>@lorawan_handler:join:117 Join-Accept in RX2: {rxq,867.1,<<"SF8BW125">>,<<"4/5">>,{{2019,7,8},{6,40,17.757036}},undefined,314341244,-12,9.3} {0,0,869.525}
2019-07-08 14:41:03.012 [info] <0.4207.0> device BE7A00000000031B {join,<<"AD7325D3">>}
2019-07-08 14:41:03.012 [warning] <0.4207.0> node AD7325D3 {repeated_reset,612}
2019-07-08 14:41:03.012 [debug] <0.4207.0>@lorawan_mac:encode_accept:441 Join-Accept <<"AD7325D3">>, netid <<18,52,86>>, cflist undefined, rx1droff 0, rx2dr 0, appkey <<"A3B79F51F639B2B43460707D2AA86D4A">>, appnce <<"E5EE24">>
2019-07-08 14:41:03.013 [debug] <0.4207.0>@lorawan_handler:join:114 Join-Accept in RX1: {rxq,867.1,<<"SF12BW125">>,<<"4/5">>,{{2019,7,8},{6,41,2.38218}},undefined,358945420,-57,-7.8}
2019-07-08 14:41:22.756 [info] <0.4217.0> device BE7A00000000031B {join,<<"AD7325D3">>}
2019-07-08 14:41:22.756 [warning] <0.4217.0> node AD7325D3 {repeated_reset,613}
2019-07-08 14:41:22.757 [debug] <0.4217.0>@lorawan_mac:encode_accept:441 Join-Accept <<"AD7325D3">>, netid <<18,52,86>>, cflist undefined, rx1droff 0, rx2dr 0, appkey <<"A3B79F51F639B2B43460707D2AA86D4A">>, appnce <<"55463B">>
2019-07-08 14:41:22.757 [debug] <0.4217.0>@lorawan_handler:join:117 Join-Accept in RX2: {rxq,867.3,<<"SF7BW125">>,<<"4/5">>,{{2019,7,8},{6,41,22.082765}},undefined,378665667,-11,8.5} {0,0,869.525}
2019-07-08 14:41:34.445 [info] <0.4222.0> device BE7A00000000031B {join,<<"AD7325D3">>}
2019-07-08 14:41:34.445 [warning] <0.4222.0> node AD7325D3 {repeated_reset,614}
2019-07-08 14:41:34.445 [debug] <0.4222.0>@lorawan_mac:encode_accept:441 Join-Accept <<"AD7325D3">>, netid <<18,52,86>>, cflist undefined, rx1droff 0, rx2dr 0, appkey <<"A3B79F51F639B2B43460707D2AA86D4A">>, appnce <<"AC2311">>
2019-07-08 14:41:34.446 [debug] <0.4222.0>@lorawan_handler:join:114 Join-Accept in RX1: {rxq,867.1,<<"SF8BW125">>,<<"4/5">>,{{2019,7,8},{6,41,33.828604}},undefined,390411916,-11,9.3}
2019-07-08 14:42:26.319 [info] <0.4264.0> device BE7A00000000031B {join,<<"AD7325D3">>}
2019-07-08 14:42:26.319 [warning] <0.4264.0> node AD7325D3 {repeated_reset,615}
2019-07-08 14:42:26.320 [debug] <0.4264.0>@lorawan_mac:encode_accept:441 Join-Accept <<"AD7325D3">>, netid <<18,52,86>>, cflist undefined, rx1droff 0, rx2dr 0, appkey <<"A3B79F51F639B2B43460707D2AA86D4A">>, appnce <<"958963">>
2019-07-08 14:42:26.320 [debug] <0.4264.0>@lorawan_handler:join:117 Join-Accept in RX2: {rxq,867.3,<<"SF12BW125">>,<<"4/5">>,{{2019,7,8},{6,42,25.698913}},undefined,442254220,-55,-8.5} {0,0,869.525}
2019-07-08 14:42:38.293 [info] <0.4269.0> device BE7A00000000031B {join,<<"AD7325D3">>}

server sens join ack on different freq and DR?
I adjusting RX1/RX2 time little to check ack. but could not get it.
Any advice?
Best Regards!

gotthardp, 2019-07-08:
There is a note about this in a troubleshooting guide
https://github.com/gotthardp/lorawan-server/blob/master/doc/Troubleshooting.md#repeated_reset

I cannot say much more abut this.

arigus, 2019-07-09:
understud, thanks

closed 2019-07-09 by arigus