gotthardp/lorawan-server #652: [Question] How can we detect the rx time per gateway uplink?

kcyung3663, 2019-07-17:
It is known that the server will *combine* the rxpk into one, but what if I want to detect the tmms of rxpk per gateway? Any solution to this scenario? Could editing the lorawan_gw_forwarder.erl and make it up be one?

gotthardp, 2019-07-17:
I am not sure what you mean. The server *never* combines two rxpk into one.

kcyung3663, 2019-07-17:
Sorry for that. What I have observed so far is, per packet sent via gateway to network server, will be outputted as one json datum, i.e. the one we will received in websocket interface(after registering handler).

In the datum, `all_gw` can be enabled to show. And my question is, how can I add the rx time into `all_gw` for each of its children?

gotthardp, 2019-07-17:
Both `time` and `tmms` is (in the latest version) already included in the `all_gw`. If you don't see it, then your packet_forwarder does not send it (usually) because GPS signal is missing and precise timing is not available.

kcyung3663, 2019-07-17:
Ooops, sorry I haven't checked it out. Will get back to you some days later. Plz hang this issue for a while.

Many thanks! ;)

kcyung3663, 2019-07-19:
it turns out still the same result as in 0.6.x, plz see attached.

Is there anything I've set wrong?

gotthardp, 2019-07-19:
Could you please have a look (and post here) what the gateways send? The packet_forwarder's JSON sent to the server.

kcyung3663, 2019-07-19:
Here is the content of `Data`:

```PÂ`Å¨ÿþou0{"rxpk":[{"tmst":571910684,"chan":1,"rfch":0,"freq":923.400000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":9.8,"rssi":-45,"size":56,"data":"QEUjwauA7wkCCXLuFmhmlu95z/amSibNz3I6YMN1DoAjeQenewhHZNgsv/0Yjn1CnQP9bQlcRcw="}]}```

gotthardp, 2019-07-19:
This is the problem with `tmms`: there is none in the rxpk. The server just uses whatever date the gateway sent. There is no date nor tmms in that rxpk, so the server sent nothing.

kcyung3663, 2019-07-19:
You are right. What if I wanna include also `tmst`? That'd make editing the lorawan_gw_forwarder.erl the only way right? lol

gotthardp, 2019-07-19:
`tmst` is on every gateway different. I don't think it is useful for anything other than scheduling downlinks. Or?

kcyung3663, 2019-07-19:
My project team is current trying to do localization with LoRa, and we want to take some timestamp data for reference. Given that the project site has no GPS coverage (super weak signal).  

Hence, we want to get at least `tmst` for latency reference, even though its precision is of second level.  

One more thing we don't understand is, if `tmms` is time from GPS, then where does `time` come from?  
If both of `time` and `tmms` are from GPS, why would they co-exist?  
If `time` is not from GPS, then does this mean we can still get a chance to obtain `time` under the circumstance of no GPS signal?

gotthardp, 2019-07-19:
The `time` provided by the gateway is UTC time and `tmms` is a GPS time, that's probably the reason to have both. But both come from the GPS signal. If the gateway does not provide anything (which is your case) the server provides its own UTC time instead.
I am afraid that without precise timing (GPS) you cannot do localization. The `tmst` is an internal number, which doesn't help you because it is different on every gateway. How would you use it?

kcyung3663, 2019-07-24:
Right. So after looking deeper, we found that the only possible way to do could be patching a modified version of packet forwarder to the gateway.

gotthardp, 2019-07-24:
I still doubt the clock precision will be sufficient, but I bet you considered this already. Shall I close this issue then?

kcyung3663, 2019-07-24:
yes plz, but one last thing: may I have your confirm that in v0.6.7, `tmms` and `time` are included in `all_gw`?

Many thanks ;)

gotthardp, 2019-07-24:
Yes. Both `time` and `tmms` are included (when provided by the gateway) in `all_gw` since v0.5.4, so it's also in your v0.6.7.

closed 2019-07-24 by gotthardp