gotthardp/lorawan-server #654: fcnt increase to 65535 get error:bad_mic

gaohcGY, 2019-07-24:
Hi, 

I'm used version is 0.6.4

in profile,the "FCnt Check" is Disabled,when device's FCnt increase to 65535,server get error "bad_mic".

32-bits frame counter is used for device.

my setting is error? or I should to update my server version of install?
[image]
[image]

Thandks
Regards

gotthardp, 2019-07-24:
Hello. This is a correct behaviour. You need to enable FCnt check to count beyond 65535.

gaohcGY, 2019-07-24:
Very thanks for your's reply in busy. Can i understand that?  Because of we cannot distinguish between 16-bit and 32-bit in server, So FCnt check is used strict 16-bit by default.

gotthardp, 2019-07-24:
The server has to know whether the device has 16b or 32b FCnt counter. The Disable option is only for debugging purposes when the device behaves erroneously. You should not use the Disable mode for so long.

gaohcGY, 2019-07-24:
Oh，I understand, thank u. 

And i have an idea.
Beause of the device has 16b or 32b FCnt counter,Can we use 32b to compatible 16b? In this, the server does not need to know what type of FCnt counter.That is more simple  for use and management.

Is this idea feasible?

gotthardp, 2019-07-24:
I don't understand. Could you give me an example?

gaohcGY, 2019-07-24:
OK, thanks for your help!   I'll try it, and send the results in here.