gotthardp/lorawan-server #655: ADR Auto-Adjust will set the channels to default.

mitwave, 2019-07-30:
Hi.
When I use ADR "Auto-Adjust" mode with setting "Set Channels" = "0-6" in AS923 (Japan), even if device got CFList which is contain correct frequency list and DR range, server will reset "used channels" to 0-1 (default join channels).
So, after receiving 1st "Link ADR MAC Command" after join, device usable channels will be changed from 0-6 to 0-1...

However, if I do...
1. set "Maintain" mode and set "Set Channels" = "0-6"
2. Server will send "Link ADR Command" and device will use 0-6.
3. After step 2, I change the "ADR mode" from "Maintain" to "Auto-Adjust".
device continues to use 0-6...

How can I use ADR Auto-Adjust mode with auto DR/Power and channel control?

altishchenko, 2019-07-30:
Where are ADR mode and Set Channels configure? Are they in the same place or in different ones, like profile ADR tab and device's ADR tab? We use this setting on the profile level and never had any issues, given the Network settings are correct. Can you provide your network/profile/device settings screenshots?

mitwave, 2019-07-31:
Hi althischenko.
Networks -> General and ADR are set based on this doc. (https://github.com/gotthardp/lorawan-server/blob/master/doc/Regions.md)
And ADR setting on device, I never change parameters.
[image]

ADR on Profile settings:
[image]

I guess, you can see the difference between ADR "Set Channels" setting on profile and ADR "Used Channels" status on device...
I don't understand why there is difference of channels...

altishchenko, 2019-07-31:
Do you have your extra channels configured in your Infrastructure -> Networks -> Channels tab? You should have them added there. Here is an example for the EU868 (notice, the channels range is set to the full value, but only the EXTRA channels are configured):
[image]

mitwave, 2019-08-01:
Hi althischenko.
Yes, I already set extra channels on "Networks" -> Channels Tab.
Before receiving "Link ADR Command", my device is using all chs (default join ch (2ch) + extra ch (5ch) via CFlist) but after receiving the command, my device is using only default join ch...

mitwave, 2019-08-01:
I have one question.
>notice, the channels range is set to the full value, but only the EXTRA channels are configured
Does this mean I should set 0-7 (or 0-6) on "Initial Channels"?
In my understanding, "Initial Channels" = "Join Channels".
So, on the AS923, my setting is... "Join Channels" = "0-1" = "Initial Channels"...
Is this OK for AS923?

altishchenko, 2019-08-01:
Hmm, now to think of it.. I am a bit confused. I never actually bothered setting Set Channels in the profile and device sections, but my Initial Channels state the full specter (0-7) and per device Used Channels line confirms that it is ok. Can you have a look into the debug.log searching for link_adr_req and link_adr_ans keys? There should be a clue somewhere. Also, decision to hop a channel is a device made decision, the server can only notify of the list of available ones.

altishchenko, 2019-08-01:
Here is an example:
2019-08-01 16:03:25.055 [debug] <0.1165.7>@lorawan_mac_commands:build_fopts:77 000A0B96 <- [{link_adr_req,5,1,255,0,0}]
2019-08-01 16:03:43.449 [debug] <0.1226.7>@lorawan_mac_commands:handle_fopts:16 000A0B96 -> [{link_adr_ans,1,1,1}]
where 255 on the first line means channels from 0 to 7 inclusive (so you will have something like 127 instead)