gotthardp/lorawan-server #656: Add new generic JSON field

hallard, 2019-07-31:
I just found this one #437, this one is what I needed, thanks, really makes sense.

But on each project, I need different parameters to be passed to final app, until now I'm using app args as a coma (or other) delimited field like that `param1,param2,param3`
But it's not friendly and can make confusion on GUI. Would it be possible to add a new JSON field for that, like this we can do whatever on the other end. and this field could be for example
```
 {
  "param1" : "Value1,
  "param2" : 2,
  "name" : "A great sensor"
}
```
This would overcome the parameters limitation and could be customized on demand for each user application.

Of course we could do this on existing app args field, but I'm already using it for other purpose and may be lot of us and last I don't think it was the initial goal.

What do you think ?

gotthardp, 2019-07-31:
It makes sense to have a standard way for encoding multiple named arguments.
It does not make sense (to me) to create yet another opaque field for that. If it's an application arg, I'd use `args` for it.

altishchenko, 2019-08-01:
May be you should also consider using App Identifier field from your profile settings. It can be used as as a sort of grouping identifier and is also available in the Uplink Fields of a Handler and as a macro expansion in the Connector url/topic fields. You can use it as a device class or device mode id with args on a particular device detailing it even further. Of course, you can also use app, desc and profile fields for that purpose too.

hallard, 2019-08-01:
Ok guys, makes sense. I used JSON in app args
Anyway, would it possible that if app args is JSON type to parse it and send JSON decoded data directly on uplink field?

altishchenko, 2019-08-06:
Hmm, this doesn't sound straight forward and, what's worse, will require some way to distinguish between JSON and non-JSON appargs in the server logic (at the moment appargs is opaque - what you write is what you get). On the other hand, if you put a valid JSON into the app args field and add it to your Uplinks in the result you will get a JSON encoded subobject with the name 'appargs'.

There is another black magic, if you really really need it... but it seriously depends on your 'data' field.
If your 'data' field is just a field ("data":"lalalala"), which you parse on the server separately (I assume this from your post), then you can use Parse Uplink function to 'flatten' the resulting JSON structure :) like this:
```
fun (Fields, _) ->
    AppArgs = jsx:decode(maps:get(<<"appargs">>, Fields)), % Decompose JSON encoded appargs into a Map
    maps:merge(Fields, AppArgs)   % Resulting Map is a merge between the two
end.
```
This way you will have your appargs contents added to the uplink JSON.

altishchenko, 2019-08-06:
Corrected version of the code above, tested on a live sever:
```
fun(Fields, _) ->
    case maps:get(appargs, Fields, none) of
        none ->
            Fields;
        Value ->
            AppArgs = jsx:decode(Value, [return_maps]),
            maps:merge(Fields, AppArgs)
    end
end.
```
With appargs given as `{"letitbe":"song"}` results in a uplinks JSON:
```
{
    "appargs":"{\"letitbe\": \"Song\"}",
    "appid":"gollum",
    "data":"00010502B50102050154",
    ...
    "letitbe":"Song"
}
```