gotthardp/lorawan-server #659: The lorawan-server HTTP output stops when the client is restarted

hjojohan, 2019-08-09:
Hi 
I experience trouble at time when my httplistner client needs to be restarted. The lorawan server runs in the background and I can se sensorframes arriving but there is no data pushed out to my httplistner software http://localhost:3100/ 

Then if a press "save changes" button in the "edit connector" dialog the sensordata arrives in my httpclient and runs fine until next time i need to do a restart of the client.

Seems like the lorawan server gives up trying to send data over HTTP if the client software is not running for a while.  It makes a autostart scenario difficult to achive 

Is there anyone out there that have a solution for this behaviour ?  thanks in advance :-)

gotthardp, 2019-08-09:
Hello. Two questions first:
 * I suppose there is no Failed flag in your connector, right?
 * Are there some errors and/or warnings in the log regarding to this?

hjojohan, 2019-08-09:
The Lorawan server shows 'connecting' flag, but it never reconnects and my application dont recieve data unless I manually press "save changes" button, then the data is flowing again. 

My http client starts, opens the reciving port and wait for json data to arrive from the lorawan server 
[image]
[image]

hjojohan, 2019-08-09:
Where can i find the log file and what is the filename ?

gotthardp, 2019-08-09:
Depending on your installation the server logs are stored in `/var/log/lorawan-server`,
`/var/lib/lorawan-server/log` or just `lorawan-server/log`. By default three log
files are provided: `debug.log`, `error.log` and `crash.log`.

saini-jitendra, 2021-09-13:
Hello @hjojohan ,

Did u resolve the issue as I m also facing the same problem?

if yes, could you please share snapshots of the Connector and handler tab?  

Data is arriving on the Lorawan server, but there is no data pushed out to my HTTP listener software.