gotthardp/lorawan-server #660: Registering a range of device addresses for ABP

mwerensteijn, 2019-08-13:
Hi,

Currently we have been using your network server very succesfully for the past two years.
We have nodes that send their status updates over Lora.
And we use access-by-personalization because we have a large amount of nodes per gateway and the nodes can be out of range.
So we don't want to wait for the nodes to join the network and we don't want the gateway to send so many join packets to nodes.

The data we send is not confidential, that is why we hardcoded the same appskey and nwkskey in the firmware for ease of use.

To receive our data over MQTT, we would need to register many nodes (10000? 20000? 40000?) individually within the Lorawan server.
We could automate that using the REST API.
However, it would be awesome if we could register a range of device addresses with the same appskey and nwkskey.

Can this be implemented?
Or do you have any suggestions on how to tackle this?

Thanks in advance!

Kind regards,
Mitchell

gotthardp, 2019-08-13:
Hello. I'd prefer a solution that could be reusable by other users. For example, some kind of a provisioning callback (Erlang fun) that would be called when a frame from an unknown device arrives. This would enable the user to implement a custom logic to ignore, log the frame, or create a device with custom generated (or statically preconfigured) keys.

mwerensteijn, 2019-08-13:
Thank you for your quick response! That sounds like a good solution

mwerensteijn, 2019-08-13:
As I have a lack of knowledge of where to add the provisioning callback, could you implement this? If not, I will try it myself but I will need to dive in Erlang and the source code. Thanks in advance