gotthardp/lorawan-server #661: Windows installation

bobcrichton, 2019-08-14:
I am trying to set up for Windows Server.  When I try to install as a service using CMD I get the following response:

Can you please advise?, Thanks

[image]

gotthardp, 2019-08-14:
It's the quotation marks and space-delimited names. Again. It could be similar to #508 or not, so the fix in https://github.com/gotthardp/lorawan-server/commit/fc80f8f56f9c9837f66f6f55d3a9de40041081b7 might help or not. It seems to be different on every Windows machine, not sure why.

gotthardp, 2019-08-14:
Would you be able to use the try-and-error approach and insert `"` or `\"` somewhere to fix it? I can give you some suggestions, but you may be faster without my help.

bobcrichton, 2019-08-14:
The fix in #508  set FILES=!FILES! \"%%A\ebin\" did the trick - thanks.

bobcrichton, 2019-08-21:
Multitech AP gateway has connected to server OK.  However as soon as I attach a device (Elsys sensor) gateway status shows "dwell time violated" and the dashboard shows repeated "join" and "ack lost".  Node page shows device is resetting frequently and no RX packets are shown.

I am moving across from The Things Network where everything worked OK so I know the devices and gateway are good, so I suspect I have missed something in the setup.

Being a private network, am I allowed to invent my own AppEUI and AppKey?  Must these be different for each device?  Are there any restrictions other than HEX length?

Finally there is nothing showing in the server log files.

Regards, Bob

gotthardp, 2019-08-21:
I suspect the downlink is not configured properly. See the Troubleshooting guide: https://github.com/gotthardp/lorawan-server/blob/master/doc/Troubleshooting.md#no-downlink-frames-delivered

The AppEUI and AppKey must match whatever is configured in the devices, but no restrictions other than the HEX length apply.

bobcrichton, 2019-09-03:
Thanks for your prompt reply.  I have only just been able to get back onto this and am still struggling.

I have checked through all the steps in the troubleshooting guide and cannot find any errors in the set-up.  However the log files are full of messages that I cannot interpret.  For example this occurs every few seconds in the crash log:
----------------------
_2019-09-03 16:33:49 =ERROR REPORT====
** Generic server lorawan_gw_forwarder terminating 
** Last message in was {udp_error,#Port<0.421>,econnreset}
** When Server state == {state,#Port<0.421>,#{}}
** Reason for termination ==
** {function_clause,[{lorawan_gw_forwarder,handle_info,[{udp_error,#Port<0.421>,econnreset},{state,#Port<0.421>,#{}}],[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_forwarder.erl"},{line,50}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,637}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,711}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}
2019-09-03 16:33:49 =CRASH REPORT====
  crasher:
    initial call: lorawan_gw_forwarder:init/1
    pid: <0.2534.0>
    registered_name: lorawan_gw_forwarder
    exception error: {function_clause,[{lorawan_gw_forwarder,handle_info,[{udp_error,#Port<0.421>,econnreset},{state,#Port<0.421>,#{}}],[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_forwarder.erl"},{line,50}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,637}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,711}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}
    ancestors: [<0.2532.0>,<0.329.0>,<0.297.0>]
    message_queue_len: 0
    messages: []
    links: [<0.2532.0>,#Port<0.421>]
    dictionary: []
    trap_exit: false
    status: running
    heap_size: 1598
    stack_size: 27
    reductions: 4403
  neighbours:
2019-09-03 16:33:49 =SUPERVISOR REPORT====
     Supervisor: {<0.2532.0>,lorawan_gw_sup}
     Context:    child_terminated
     Reason:     {function_clause,[{lorawan_gw_forwarder,handle_info,[{udp_error,#Port<0.421>,econnreset},{state,#Port<0.421>,#{}}],[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_forwarder.erl"},{line,50}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,637}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,711}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}
     Offender:   [{pid,<0.2534.0>},{id,packet_forwarder},{mfargs,{lorawan_gw_forwarder,start_link,[[{port,1680}]]}},{restart_type,permanent},{shutdown,5000},{child_type,worker}]_
------------------
There are also lots of entries in the debug and error logs.

Does this offer any clues as to what is wrong in my set-up?

Thanks, Bob

gotthardp, 2019-09-03:
The root-cause is `udp_error econnreset`. That is a quite unusual network issue. Is (or was recently) the port 1680 used? Or are you trying some unusual setup/activity?

bobcrichton, 2019-09-03:
Thanks for your prompt reply, Petr

Here is the gateway_conf:

"gateway_conf": {
                                  "gateway_ID": "<WILL-BE-AUTO-REPLACED-WITH-LORA-EUI>",
                                  "server_address": "82.113.131.209",
                                  "serv_port_up": 1680,
                                  "serv_port_down": 1680,
                                  "keepalive_interval": 10,
                                  "stat_interval": 30,
                                  "push_timeout_ms": 100,
                                  "forward_crc_valid": true,
                                  "forward_crc_error": true,
                                  "forward_crc_disabled": false,
                                "synch_word": 52,
                "autoquit_threshold": 60
                }

The sys.config file on the server is as downloaded.

As far as I am aware, I have set up a standard configuration.  I have asked our server provider to ensure that the server at 82.113.131.209<http://82.113.131.209/> has UDP port 1680 fully open.  I had assumed that the gateway connecting OK confirmed this was so.

Regards, Bob

gotthardp, 2019-09-03:
And when exactly this error occurs?

bobcrichton, 2019-09-03:
All sensors switched off and server restarted.  Here is an initial timeline.

Debug log shows:

2019-09-03 22:50:41.169 [debug] <0.122.0>@lager_handler_watcher:126 Lager installed handler {lager_file_backend,"error.log"} into lager_event

2019-09-03 22:50:41.169 [debug] <0.128.0>@lager_handler_watcher:126 Lager installed handler error_logger_lager_h into error_logger

2019-09-03 22:50:41.653 [debug] <0.117.0>@lager_handler_watcher:126 Lager installed handler lager_backend_throttle into lager_event

2019-09-03 22:50:41.778 [debug] <0.309.0>@lorawan_app:start:17 Using config: [{max_lost_after_reset,10},{gateway_delay,200},{frames_before_adr,50},{slack_server,{"slack.com",443}},{applications,[{<<"semtech-mote">>,lorawan_application_semtech_mote}]},{http_admin_path,<<"/admin">>},{websocket_timeout,3600000},{devstat_gap,{432000,96}},{trim_interval,3600},{http_extra_headers,#{}},{deduplication_delay,200},{connectors,[{lorawan_connector_amqp,[<<"amqp">>,<<"amqps">>]},{lorawan_connector_mqtt,[<<"mqtt">>,<<"mqtts">>]},{lorawan_connector_http,[<<"http">>,<<"https">>]},{lorawan_connector_mongodb,[<<"mongodb">>]},{lorawan_connector_ws,[<<"ws">>]}]},{event_lifetime,86400},{packet_forwarder_listen,[{port,1680}]},{http_admin_redirect_ssl,true},{http_admin_credentials,{<<"admin">>,<<"bobc1911">>}},{server_stats_interval,60},{retained_rxframes,50},{ssl_options,[]},{http_admin_listen,[{port,8080}]},{http_admin_listen_ssl,[{port,8443},{certfile,"cert.pem"},{cacertfile,"cacert.pem"},{keyfile,"key.pem"}]},{http_custom_web,[]}]

2019-09-03 22:50:41.778 [warning] <0.309.0>@lorawan_app:retrieve_valid_ssl:46 http_admin_listen_ssl not configured

2019-09-03 22:50:41.794 [info] <0.341.0> server lorawan@WIN-IB6VT3AF093 started

2019-09-03 22:50:41.810 [debug] <0.349.0>@lorawan_http_registry:update_routes:62 New routes []

2019-09-03 22:50:47.591 [warning] <0.357.0>@lorawan_mac:ingest_frame0:62 gateway 008000000001562F received bad frame (mtype 111): <<252,153,91,132,4,145,210,139,22,146,53,23,151,82,85,63,251,78,247,70,97,62,61,84,21,70,164,238,161,153,160,177,121,141,243,189,114,145,45,218,79,152,69,27,229>>

2019-09-03 22:50:48.153 [info] <0.346.0> gateway 008000000001562F {connected,{{212,159,114,126},35367}}

2019-09-03 22:50:55.263 [warning] <0.361.0>@lorawan_mac:ingest_frame0:62 gateway 008000000001562F received bad frame (mtype 110): <<208,66,17,46,72,213,243,39,221,87,70,253,6,25,242,52,165,232,73,9,78,163,211,4,85,18,150,45,57,78,83,40,47,79,152,134,132,37,14,142,104,73,161,109,189,17,193,115,205,77,207,186,205,66,248,206,137,130,74,75,56,5,53,168,31,132,68,28,122,43,240,128,156,208,151,43,137,81,185,159,236,98,118,93,94,31,25,122,234,45,28,221,81,146,56,137,232,1,18,64,239,9,199,23,86,198,239,89,78,254,213,65,113,233,45,27,41,37,62,252,100,38,210,177,16,180,214,91,40,21,16,151,21,95>>

2019-09-03 22:51:02.794 [warning] <0.346.0> gateway 008000000001562F {ack_lost,100.0}

2019-09-03 22:52:46.671 [warning] <0.392.0>@lorawan_mac:ingest_frame0:62 gateway 008000000001562F received bad frame (mtype 0): <<0,248,181,201,127,161,154,83,12,238,65,16,8,49,153,181,197,89,235,219,216,131,192,213,105,120,58,207,115,76,162,76,230,96,35,207,238,92,250,230,167,40,193,109,208,168,193,164,56,106,29,104,109,47,185,180,188,236,202,102,240,45,53,55,177,7,216,29,224,149,37,255,112,62,86,168,19,123,40,243,49,4,196,118,165,206,53,102,154,240,9,114,185,188,174,5,39,109,237,152,105,47,225,101,235,205,189,66,126,191,61,21,59,188,207,200,241,66,46,67,48,22,211,67,207>>

2019-09-03 22:53:02.781 [warning] <0.346.0> gateway 008000000001562F {ack_lost,100.0}

2019-09-03 22:54:03.813 [warning] <0.343.0>@lorawan_db_guard:send_alert:201 gateway 008000000001562F [{<<"disconnected">>,2}] [{<<"dwell_time_violated">>,13}]

2019-09-03 22:54:06.907 [info] <0.346.0> gateway 008000000001562F {connected,{{212,159,114,126},59464}}

2019-09-03 22:54:06.938 [info] <0.347.0>@lorawan_gw_forwarder:terminate:128 packet_forwarder interface terminated: {function_clause,[{lorawan_gw_forwarder,handle_info,[{udp_error,#Port<0.11>,econnreset},{state,#Port<0.11>,#{}}],[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_gw_forwarder.erl"},{line,50}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,637}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,711}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}

2019-09-03 22:54:06.938 [error] <0.347.0>@lorawan_gw_forwarder:handle_info:50 gen_server lorawan_gw_forwarder terminated with reason: no function clause matching lorawan_gw_forwarder:handle_info({udp_error,#Port<0.11>,econnreset}, {state,#Port<0.11>,#{}}) line 50

2019-09-03 22:54:06.938 [error] <0.347.0>@lorawan_gw_forwarder:handle_info:50 CRASH REPORT Process lorawan_gw_forwarder with 0 neighbours crashed with reason: no function clause matching lorawan_gw_forwarder:handle_info({udp_error,#Port<0.11>,econnreset}, {state,#Port<0.11>,#{}}) line 50

2019-09-03 22:54:06.938 [error] <0.345.0>@lorawan_gw_forwarder:handle_info:50 Supervisor {<0.345.0>,lorawan_gw_sup} had child packet_forwarder started with lorawan_gw_forwarder:start_link([{port,1680}]) at <0.347.0> exit with reason no function clause matching lorawan_gw_forwarder:handle_info({udp_error,#Port<0.11>,econnreset}, {state,#Port<0.11>,#{}}) line 50 in context child_terminated

[... cut ...]
2019-09-03 23:05:11.105 [error] <0.597.0>@lorawan_gw_forwarder:handle_info:50 CRASH REPORT Process lorawan_gw_forwarder with 0 neighbours crashed with reason: no function clause matching lorawan_gw_forwarder:handle_info({udp_error,#Port<0.57>,econnreset}, {state,#Port<0.57>,#{}}) line 50

2019-09-03 23:05:11.105 [error] <0.345.0>@lorawan_gw_forwarder:handle_info:50 Supervisor {<0.345.0>,lorawan_gw_sup} had child packet_forwarder started with lorawan_gw_forwarder:start_link([{port,1680}]) at <0.597.0> exit with reason no function clause matching lorawan_gw_forwarder:handle_info({udp_error,#Port<0.57>,econnreset}, {state,#Port<0.57>,#{}}) line 50 in context child_terminated

gotthardp, 2019-09-04:
There are three indicators:
* ECONNRESET error, which someone said *Windows seems to use this error code to indicate that the socket passed to recvfrom previously sent some traffic to a local address which was not listening. A superficial reading of POSIX suggests that ECONNRESET is indeed a valid return for recvfrom().*
* Gateway reconnect (change of gateway port) from 35367 to 59464.
* 100% ACK lost

I suspect the gateway is behind a NAT that does not maintain a stable port mapping towards the server in public internet and doesn't allow incoming UDP traffic (from server to the gateway). You need a [masquerade with stable port mapping](https://www.ibm.com/support/knowledgecenter/en/ssw_ibm_i_72/rzajb/rzajbrzajb4dportnat.htm).

bobcrichton, 2019-09-04:
The gateway is behind a NAT with the gateway static internal IP mapped to the public static IP address along with UDP port 1680. All worked fine on The Things Network (port 1700) so I am a little puzzled.

I have asked our server provider also to investigate and will get back to you as soon as this is complete.

Thanks again for your prompt response.

bobcrichton, 2019-09-04:
It turned out that the UDP port was not opened for the IP address being used on the server for communications with the gateway. All much calmer now with no crashes, etc. with two sensors attached.

All I need to do now is write a backend to pass received data to MS SQL Server (on the same physical server).  Any suggestions where to start?

Thanks for your patience with a novice!

Apaisal, 2019-09-05:
I also has encountered this issue on Windows 10 server and investigating. On linux isn't found.
[image]
[image]