gotthardp/lorawan-server #664: Integration: MQTT Broker on a non local server

eng56, 2019-08-21:
Hello, I have a server where I'm running Node Red,  I have just installed a LoriXOne gateway and I have  set up gotthardp on it. 
I'm using the MQTT broker on node red with the following server adress: localhost:1883 , when I run it locally that's working. 
When I'm switching to another server who is not on the same network I need  to change "localhost" by the server adress. How can I find it on gotthardp ? 

I'm sorry if my question is stupid but I'm really stuck, 

Thank you, 

[image]

gotthardp, 2019-08-22:
Well, the MQTT broker address is in the **Connectors** section.

eng56, 2019-08-23:
Thank you for your answer, I think my question was not clear: 
 I have a server with a websited hosted on it like: https://mydomain.com/nodered. 
My LoriXone is set up with Mosquitto and gotthardp. 
I'm trying to forward my data from my gateway to my nodered. 
I have tried with a MQTT node and also with a post request, both are not working. 
I have read and I have tried to do what you explain in your doc/Connectors  and doc/Troubeshooting

Thank you for your time 

This what I have done for POST: 

[image]

gotthardp, 2019-08-23:
And where is your nodered accessible? [This site](https://cookbook.nodered.org/http/post-json-data-to-a-flow) says it may work with HTTP and port 1880.

eng56, 2019-08-23:
My nodered is accessible via Internet, just have to browse to my http://domain.com/nodered. 
I'm already making POST with Sigfox or TTN, and it's working well. 
Thank you for your help

gotthardp, 2019-08-23:
Well, it's hard to debug remotely, but:
 * Try using `http` instead `https` as it's simpler
 * Make sure there is a `\` slash at the beginning of the **Publish Uplinks** and others (as described [here](https://github.com/gotthardp/lorawan-server/blob/master/doc/Connectors.md#httprest)
 * Make sure the port and path matches whatever is configured in your nodered.

eng56, 2019-08-23:
Yes i know, thnak you again for your help. 
. I'm using http
. It should be `/` instead of  `\` no ? 
.For the debug I have an error of  unknown_deveui, it's mean that the DevEUI is not correctly configured but I don't have this error if I remove the connector and I let only the connector that is working (mqtt on local node red).  I don't understand beacause devEUI is linked to the device and not to the connectors, no ? 

Thank you 

--

gotthardp, 2019-08-23:
Yeah, `/`, sorry for the confusion.
I am lost a bit now. We should fix one error at a time. Please, get it working with MQTT (you must not see ` unknown_deveui`) and then just change the URL and the `/`.

eng56, 2019-08-23:
No problem.
With MQTT set up to a local node red it's working. This the config that is working:
[image]

When I try to make a POST on my local node-red or on another node red in a distant server it's not working with this config: (I'm letting subscribed and received topic in blank because I just want to receive data)
[image]

eng56, 2019-08-23:
I have set up my LoraW One yesterday so I think I have the last version. 
I have this error when I'm trying to make a post to my distant server (i don't have it with my local nodered):
` http_error{ 301,<<" http://mywebsite.com/nodered/public/lorawantest">>,<<"/nodered/public/lorawantest/lora/aiq/up">>}`

Yes I know how tu use wireshark, I'am on the same Wifi Network than my LoriXOne do I need to install WireShark on it or can I use it directly on my computer ? 
I'm really gratefull for your help

gotthardp, 2019-08-23:
Problem found: my server cannot handle "301" (sent by nodered) as already reported by #534 
Can nodered be convinced to not send it? :-)

eng56, 2019-08-23:
Ok thank you for your answer so it means that for the moment it's impossible to make a POST from Gotthardp to node red ?

gotthardp, 2019-08-23:
Unless nodered can be convinced to not send it.

eng56, 2019-08-23:
Ok thank you again for your help and your time

closed 2019-08-23 by eng56

gotthardp, 2019-08-23:
I hope that you have server 0.6.x, your HTTP connector is *Enabled* and there is no *Failed* flag. Right?
Is there anything in the nodered/server logs showing a HTTP error? Would you be able to operate https://www.wireshark.org/ to see what HTTP traffic is sent where?