gotthardp/lorawan-server #665: How toconnect my site with lorawan websocket?

Papa005, 2019-08-23:
Hello! I need to make a site where it will be displayed, what is sent to the lorawan server by the device. How to do it?
[image]
I already wrote this in script.js file
`let socket = new WebSocket("ws://5.134.219.10:48080/ws/event");`
what's after?

gotthardp, 2019-08-23:
Have a look at https://github.com/gotthardp/lorawan-server/blob/master/priv/admin/ws.html

Papa005, 2019-08-23:
I want to do something like this 
[https://github.com/gotthardp/lorawan-server/issues/474](url)
watch his site

Papa005, 2019-08-23:
> Have a look at https://github.com/gotthardp/lorawan-server/blob/master/priv/admin/ws.html

ok

Papa005, 2019-08-23:
Can you help me how to use it?
My connectors settings is:
[image]
And.. Sorry for my English))

Papa005, 2019-08-23:
i mean how to use your script.

gotthardp, 2019-08-23:
Have you read https://github.com/gotthardp/lorawan-server/blob/master/doc/Connectors.md#web-sockets?

Papa005, 2019-08-23:
Ok, thanks. I will read. And where can i take this file? 
`<script src="/admin/jquery.min.js"></script>`

gotthardp, 2019-08-23:
Aaah. Try `<script src="https://code.jquery.com/jquery-1.12.4.min.js"></script>` instead, or simply download it from there. Or anywhere. It's a standard jquery package.

gotthardp, 2019-08-23:
Or you can use any websocket example, e.g. this: https://www.npmjs.com/package/jquery-simple-websocket

Papa005, 2019-08-23:
So, i have button, which send pocket like this to server. 
[image]

> 025BD30A3BA501 

Last numeral is type of depression. (1 - once, 2 - long press, 3 - double press).
I need simple site, which will show when i press and type of depression. 
What comand do i need?
Thank you)
Or send me some instruction of this.
My site is  [https://websockettest1.000webhostapp.com/](url)

gotthardp, 2019-08-23:
Well, this is a standard JavaScript coding question. I suggest you check stackoverflow or any other coding site.

Papa005, 2019-08-26:
Hello! I wrote to stackoverflow, but i didn't took answer. I think, only you can help me. So i have a little progress. Code HTML has: 
`<script src="script.js"></script>`
script.js code:
`let socket = new WebSocket("ws://5.134.219.10:48080/ws/nodes/FF09E1E2");`
`ws.onopen = function() { alert("Connection opened...") };`
And in 

> Events 

[image]
There is my button:
[image]

Papa005, 2019-08-26:
Also i checked this code: 
`let socket = new WebSocket("ws://5.134.219.10:48080/ws/uplink");`
`socket.onopen = function(e) {
  alert("[open] Connected");
};`
And it worked, because when i opened my site, i took js messege 

> [open] Connected

But, how to tale info about connection of my button?

Papa005, 2019-08-26:
Now, i checked this code, 
`let socket = new WebSocket("ws://5.134.219.10:48080/ws/uplink");`
`socket.onopen = function(e) {`
  `alert("[open] Connected");`
  `alert("Sending messege");`
  `socket.send("Hello");`
`};`

And i took it:
[image]
How i understood, i need to send some json command to the server, but what command?

Papa005, 2019-08-26:
May be i need to do anything here?
[image]

Papa005, 2019-08-26:
I DID IT. EEEEE. VERY THANK YOU!! 
[image]
Now i need to write this in html.

gotthardp, 2019-08-26:
Cool. I am glad it works.

closed 2019-08-26 by gotthardp