gotthardp/lorawan-server #666: MQQT uri not accepted

berrySkies, 2019-08-26:
Firstly thank you for your product.

Trying to create a new connector to an existing MQQT server in the web admin interface. "mqqt" is not accepted in the URI field (i.e. trying to enter mqqt://host:1883). Message received is "invalid form" and the URI field is highlighted in red. 

Also cannot find mailing list in google groups. Groups reports that if list exists I will need invite from owner. Is the mailing list still used?

gotthardp, 2019-08-26:
Hello. `mqqt` is not supported. If you mean `mqtt` please send me exactly what URI is not accepted.

berrySkies, 2019-08-26:
OMG! Sorry I can't believe did this. Been using MQTT for years.

closed 2019-08-26 by berrySkies

gotthardp, 2019-08-26:
That's OK :-)