gotthardp/lorawan-server #669: Gateway not connected / no downlink sent

BruceFeuillette, 2019-09-06:
Hello !

We can't have our gateway to be in the connected state. I've [read this thread](https://github.com/gotthardp/lorawan-server/issues/435) without finding a solution.
**Gateway hardware** : Raspberry Pi + Dragino HAT.
**Gateway software** : Raspbian Buster + Dragino dual channel packet forwarder or single channel packet forwarder
**Lorawan-server hardware** : VM on vSphere
**Lorawan-server software** : Debian 9 + lorawan-server 0.6.2
We are on our private network, so no firewall on lorawan-server and gateway, no NAT.

[image]

The gateway is sending stats and uplink to the lorawan-server

>stat update: 2019-09-06 07:20:05 GMT no packet received yet
> _And so on_
>CE0 Packet RSSI: -66, RSSI: -91, SNR: 9, Length: 23 Message:'...........03d4~..O..."'
> rxpk update: {"rxpk":[{"tmst":1504846727,"freq":868.3,"chan":0,"rfch":0,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","rssi":-66,"lsnr":9.0,"size":23,"data":"AKqqqgAAAQAAspowM2Q0fg7nT7CE1iI="}]}
> stat update: 2019-09-06 07:26:05 GMT 2 packets received

 The packets are received by the lorawan-server (the packets are not the same on both side, I did a copy paste

> 09:20:05.318759 IP 172.30.3.49.36346 > srv-lorawan.ville-macon.fr.1700: UDP, length 213
> E...c.@.@.x`...1...........h..F..'....m.{"stat":{"time":"2019-09-06 07:20:05 GMT","lati":0.0,"long":0.0,"alti":10,"rxnb":0,"rxok":0,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"your name","mail":"a@b.c","desc":"Dual chann  el pkt forwarder"}}
> 09:20:05.318995 IP srv-lorawan.ville-macon.fr.1700 > 172.30.3.49.36346: UDP, length 4
> E.. ..@.@..p.......1......^...F.
> ^C

So, when one of our node want to join, the server didn't send the downlink because the gateway is down.

> 2019-09-06 09:26:02.718 [info] <0.822.0> device XXXXXXXXXXX {join,<<"02CCC35D">>}
> 
> 2019-09-06 09:26:02.719 [warning] <0.822.0> node 02CCC35D {repeated_reset,271}
> 
> 2019-09-06 09:26:02.719 [debug] <0.822.0>@lorawan_mac:encode_accept:441 Join-Accept <<"02CCC35D">>, netid <<0,0,1>>, cflist undefined, rx1droff 0, rx2dr 0, appkey <<"XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX">>, appnce <<"XXXXX">>
> 
> 2019-09-06 09:26:02.719 [debug] <0.822.0>@lorawan_handler:join:117 Join-Accept in RX2: {rxq,868.3,<<"SF7BW125">>,<<"4/5">>,undefined,undefined,1504846727,-66,9.0} {0,0,869.525}
> 
> 2019-09-06 09:26:02.719 [warning] <0.342.0>@lorawan_gw_router:handle_cast:104 Downlink request ignored. Gateway B827EBFFFFB96DE6 not connected.
> 
> 2019-09-06 09:26:05.318 [warning] <0.342.0> gateway B827EBFFFFB96DE6 {uplinks_lost,1}`

We don't know what to do or where to look. :)

gotthardp, 2019-09-06:
Hello. I suspect your gateway does not send PULL_DATA frames, not sure why.
https://github.com/Lora-net/packet_forwarder/blob/master/PROTOCOL.TXT#L279

What packet_forwarder version/fork are you using?

BruceFeuillette, 2019-09-06:
This one, from Dragino : https://github.com/dragino/dual_chan_pkt_fwd
The Lora-net packet forwarder can't be used as we have a chipset SX1276 where Lora-net uses the SX1301.

gotthardp, 2019-09-06:
In the code I cannot find `PKT_PULL_DATA`, so it seems the code doesn't support downlinks

BruceFeuillette, 2019-09-06:
Yes, it seems.
I connected the gateway on resiOT and the console writes that the gateway don't support txAck.
So I guess we had to invest in a gateway with a SX1301 chip from Semtech, so more expensive !

closed 2019-09-06 by gotthardp