gotthardp/lorawan-server #671: Login wiht non default user - http 403 error

Kralg, 2019-09-09:
Hello,

With v.0.6.7 I create another user and if I try to connect with it from another browser and I am getting error 403. Is it necessary to make some additional configurations to make ti working? Server has: one device, a gateway, several frames from the end device.

[image]

altishchenko, 2019-09-10:
Don't see any login attempts in your events log. Can you try clearing the browser cache?

gotthardp, 2019-09-10:
What *scopes* do the users have?

Kralg, 2019-09-10:
If I use admin scope then I get a window below with an error message. 

[image]

[image]

gotthardp, 2019-09-10:
That's quite understandable. The user has a permission to view the web-admin pages, but has no permission to view any data. You should add more scopes depending on what data shall be accessible. Note also that the dashboard (visible bu default after login) displays an overview of everything, so the display the dashboard you need read access to all data types.

altishchenko, 2019-09-10:
Try adding server:read, network:read and device:read to the user and see if it will help. web-admin is just web interface access, but REST requests behind are :read controlled.

Kralg, 2019-09-10:
Thank you! Now it is clear and works. 
Just some funny error persists with 403 when user does not have a scope to read. Maybe it is usual, but some kind of notification - limited scope/rights, would be handy. 
I am wondering in case of older versions of server (without scopes for users) is it possible to disable web-admin view without disabling REST?

gotthardp, 2019-09-10:
Older versions cannot disable web-admin only. It's either all or nothing.

closed 2019-09-10 by Kralg