gotthardp/lorawan-server #672: HTTP downlink WebSocket fails with 500

pboettch, 2019-09-10:
Using 0.6.7 I created a WebSocket connection to test download packets:

[image]

I made my device join with and after getting its device address I ran (copied from the connection-documentation):

`curl -v -uadmin:admin -H "Content-Type: application/json" -X POST http://meli-1:8080/in/<devaddr> -d '{"data":"ABCDEF"}'`

The server returns:

```
*   Trying 192.168.1.60:8080...
* TCP_NODELAY set
* Connected to meli-1 (192.168.1.60) port 8080 (#0)
* Server auth using Basic with user 'admin'
> POST /in/578F4A58 HTTP/1.1
> Host: meli-1:8080
> Authorization: Basic YWRtaW46YWRtaW4=
> User-Agent: curl/7.65.3
> Accept: */*
> Content-Type: application/json
> Content-Length: 17
> 
* upload completely sent off: 17 out of 17 bytes
* Mark bundle as not supporting multiuse
< HTTP/1.1 500 Internal Server Error
< content-length: 0
< 
* Connection #0 to host meli-1 left intact
```

In the debug.log I can read:
```
2019-09-10 15:00:32.054 [error] <0.597.0>@lorawan_application_backend:handle_downlink:196 CRASH REPORT Process <0.597.0> with 0 neighbours crashed with reason: no match of right hand value [] in lorawan_application_backend:handle_downlink/2 line 196
2019-09-10 15:00:32.077 [error] <0.597.0>@lorawan_application_backend:handle_downlink:196 Cowboy stream 1 with ranch listener http and connection process <0.596.0> had its request process exit with reason: no match of right hand value [] in lorawan_application_backend:handle_downlink/2 line 196
```

What is my mistake?

gotthardp, 2019-09-10:
The **Application** field in your connector is empty. It's must point to some application.

pboettch, 2019-09-10:
I see... it's not yet clear to me what I have to do to create a handler for my test-case.

pboettch, 2019-09-10:
It's not clear to me what I need to put in **Build Downlinl** if I just what to send the data I receive via my POST-request as bytes to the device.

pboettch, 2019-09-10:
I created the following application: 

[image]

And when doing curl with the right devaddr `curl -v -uadmin:admin -H "Content-Type: application/json" -X POST http://meli-1:8080/in/560F4CA4 -d '{"data":"deadbeef"}'` it returns:

```
* upload completely sent off: 19 out of 19 bytes
* Mark bundle as not supporting multiuse
< HTTP/1.1 204 No Content
< content-type: text/html
< date: Tue, 10 Sep 2019 14:59:34 GMT
< server: Cowboy
< 
* Connection #0 to host meli-1 left intact
```
No error and no messages in the debug.log . But no frames seem to be sent - at least my device does not receive anything.

altishchenko, 2019-09-10:
First, you should specify 'port' attribute in your POST ("port":1). Use the number your device uses to send uplinks. Second, if it is a Class A device, you should wait for an uplink to happen, then the server will be able to send data back to it. If it is a Class C device, you should also specify "time":"immediately" in your POSTed JSON.

pboettch, 2019-09-10:
I added the port-field to my JSON. I can see the packets I send with curl listed being queued when looking at the activated device page:

[image]

My device is sending uplinks regularly but the downlink-packets are not sent.

gotthardp, 2019-09-10:
You mean the downlinks are not removed from the queue, they never appear in gateway's txpk or just that the downlink is never *received*?

pboettch, 2019-09-10:
Never removed from the queue and never appear in the packet-forwarder's log.

gotthardp, 2019-09-10:
What is on your Frames page, please? What frames related to this device you see?

pboettch, 2019-09-10:
I forced the download to go out by adding `"time":"immediately"` to my JSON message.

[image]

Then I can the txpkt in the packet-forwarder. And there is crying about the frequency. Which is set to 400 MHz. Out of range of course. I will check with my colleagues what's going on.

gotthardp, 2019-09-10:
Your application must not be semtech-mote. That one does not support custom downlinks. You need to have a custom handler and link it to the devices and the connector.

Also, immediate transmission is class C. Have a look at your netwok config and the Regions.md with recommended settings.

pboettch, 2019-09-10:
Once I changed the application in the device packets are going out and the packet-forwarder is using the right frequency. (I still don't see the incoming packet in my device, but this is probably a bug).

Thanks for helping out. It's quite hard to get the elements together.

pboettch, 2019-09-10:
Never mind, the device is receiving the message. :1st_place_medal: Thanks again.

closed 2019-09-10 by pboettch