gotthardp/lorawan-server #675: Error too early

dynawink, 2019-09-23:
In the dashboard of our LoRaWAN I see a read bubble appearing with the text 'too_early'. What is the meaning of the message? I could not find the text 'too_early' in the lorawan-server code. From where does it originate?

Cf. https://gist.github.com/dynawink/94d74ba521cb96a9971e120a0a695f99

gotthardp, 2019-09-23:
Hello. It comes from the packet_forwarder (the gateway). It might be due to https://github.com/Lora-net/packet_forwarder/issues/47, so it could effectively mean "too late".

dynawink, 2019-09-23:
Thanks for the link. It would be helpful to add this information to https://github.com/gotthardp/lorawan-server/blob/master/doc/Administration.md . The comments in https://github.com/Lora-net/packet_forwarder/blob/master/lora_pkt_fwd/src/jitqueue.c#L224 could serve as template.

JohanVikman, 2019-10-03:
I made a PR https://github.com/gotthardp/lorawan-server/pull/677

@dynawink does this look like something you would have wanted?

gotthardp, 2019-10-07:
Fixed by #677.

closed 2019-10-07 by gotthardp