gotthardp/lorawan-server #679: Handlers list

Kralg, 2019-10-08:
Hello,

I noticed one funny thing. If there is a list of handlers and if there is no handler named with a single name e.g."Handler" then the list of Applications (Handlers list) in Profile edit does not populate, but if I add handler with the name e.g. "Handler" then it appears again.

The list without a single word Handler without numbers:
[image]
No list in profile:
[image]

The list in profile populated with a single word "Handler"
[image]

gotthardp, 2019-10-08:
Hello. I have some troubles reproducing this. What exactly did you have in your Handlers list to get the erroneous handling?

Kralg, 2019-10-09:
I tried it today but was also unable to do it, I completely changed the lists from that time. 
However, the problem could come from different case. When there is a handler which is assigned to a profile and I delete this handler, then the list is not populated in the Profile edit window of which that deleted handler was used. 
In some cases it is good, as in this circumstances person can not change the handler of that Profile but in the Profile list it is possible to see what was the name of that last handler used. 
Maybe there could be some kind of notification before delete of the Handler (that this handler is being used in this profile and connector) or in Profile edit window that the assigned Handler was deleted and the name was..

gotthardp, 2019-10-09:
Oh, yes, thats a known issue (#324), which is not easy to fix.

Kralg, 2019-10-09:
I had troubles in making the Thingsboard communication over mqtt. And for quit a while I was unable to make it to run on daemon lorawan server while by manualy started server from different location it was working. 
I was getting these errors:
`
Oct 09 09:42:09 gateway lorawan-server[1252]: 09:42:09.168 [debug] Reconnecting
Oct 09 09:42:09 gateway lorawan-server[1252]: 09:42:09.171 [warning] ClientId is NULL!
Oct 09 09:42:09 gateway lorawan-server[1252]: 09:42:09.173 [info] [Client <0.460.0>]: connecting to demo.thingsboard.io:1883
Oct 09 09:42:09 gateway lorawan-server[1252]: 09:42:09.299 [info] [emqttc_gateway_4bf4796e69b68204f068@192.168.1.101:34517] SENT: CONNECT(Q0, R0, D0, ClientId=emqttc_gateway_4bf4796e69b68204f068, ProtoName=MQTT, ProtoVs
Oct 09 09:42:09 gateway lorawan-server[1252]: 09:42:09.300 [info] [emqttc_gateway_4bf4796e69b68204f068@192.168.1.101:34517] SENT: <<16,47,0,4,77,81,84,84,4,2,0,0,0,35,101,109,113,116,116,99,95,103,97,116,101,119,97,121,
Oct 09 09:42:09 gateway lorawan-server[1252]: 09:42:09.301 [info] [Client <0.460.0>] connected with demo.thingsboard.io:1883
Oct 09 09:42:09 gateway lorawan-server[1252]: 09:42:09.425 [debug] Connector <<"Conn_4th">> to <<"mqtt://demo.thingsboard.io:1883">> (attempt311) failed: {shutdown,{connack_error,'CONNACK_CREDENTIALS'}} (count: 10)
Oct 09 09:42:09 gateway lorawan-server[1252]: 09:42:09.426 [info] [Client <0.460.0>] RECV: CONNACK_CREDENTIALS
Oct 09 09:42:09 gateway lorawan-server[1252]: 09:42:09.472 [debug] Connector not available: disconnected
Oct 09 09:42:13 gateway lorawan-server[1252]: 09:42:13.208 [warning] server {http_error,{404,"/api/handlers/Handler1",<<"admin">>,"192.168.1.100"}}
Oct 09 09:42:14 gateway lorawan-server[1252]: 09:42:14.808 [warning] server {http_error,{404,"/api/handlers/Handler1",<<"admin">>,"192.168.1.100"}}
Oct 09 09:42:15 gateway lorawan-server[1252]: 09:42:15.366 [debug] Connector not available: disconnected
`
I saw that there was problem with Handler1 (which apparently was used for websocket and I deleted at some point). But my main target was mqtt with Conn_4th connector. 
Does this mean that if there is problem with any missing handler then server kills other connectors?

closed 2024-05-09 by Kralg