gotthardp/lorawan-server #680: Many many "ack lost" I have set push timeout in gateway to 500ms..

hjojohan, 2019-10-08:
Please tell me what could be wrong and how to improve my system.  I am using multitech AEP indoor GW with push timeout set to 500ms in the config-file

[image]

cocoaway, 2019-10-14:
Hi,
gateway (as I can see in your picture) is spotting  a red "!"
Have many gateways are connected to NS ?

jot4p, 2019-10-16:
This ack_lost maybe it's packages that aren't yours. Or you don't have the devices registered

closed 2019-11-03 by gotthardp

jot4p, 2019-11-04:
I have ignored "TRUE" but I receive the ack_lost packets 

[image]