gotthardp/lorawan-server #683: JSON syntax error in 1GATEWAY LORA gateway while sending Downlink

mfabdul01, 2019-11-05:
Hi i am using **1GATEWAY** LORA gateway product for the communication and i am experiencing error as JSON syntax error while i am sending Downlink Message from my plartform (using WONDERWARE ) through CloudMQTT broker. But while sending the same Topic name and message  from the MQTTbroker itself its working. how to sort out the error obtaining on the gateway?

gotthardp, 2019-11-05:
Hello. What is the JSON that you send and what is the error that you get?

mfabdul01, 2019-11-06:
HI good day ,

i had send the following message and Topic to my cloud , then the same topic name have been subscribed in my LoRa gateway ,

Topic:
in/70BD59B

Message:
{"eui":"70BD59B","data":"41544d303","port":160}

Error Message it showing is:
JSON_syntax_error

Thanks for your Replay.

gotthardp, 2019-11-06:
The EUI is too short. Also, the hex-string must have an even-number of characters, e.g. "041544d303".

mfabdul01, 2019-11-06:
hi,

the exact full eui of a device is as follows ,

in/770B3D59BA0009871
{"eui":"70B3D59BA0009871","data":"41544d303","port":160}

even with full eui it showing error

the problem is while sending the same message copied and send from the cloud its working but while doing the same from my platform it showing the error

in my case uplinks i am receiving from the sensor is fine but while send back downlink to gate its having the issue..

gotthardp, 2019-11-06:
In any case the data string is still wrong (see above). Also, please have a look in the server logs and check what errors and warnings does it show.

mfabdul01, 2019-11-07:
hi ,
i had attached my gateway configuration ,cloud data FYR
As the first data in the cloud is sent from the plartform in which it showing the error and same data i copy and send it from the cloud it is accepting..
how to access the full log information in my gateway other than events information?
[image]
[image]
[image]
[image]

mfabdul01, 2019-11-10:
HI,
while i access the server to see the error or debug information it shows the following error message,

**2019-11-10 12:59:28.027 [error] <0.7472.0>@lorawan_connector:match_vars:101 Topic <<105,110,47,55,48,66,51,68,53,57,66,65,48,48,48,57,56,55,49>> does not match pattern {{re_pattern,1,0,0,<<69,82,67,80,125,0,0,0,16,0,0,0,1,0,0,0,255,255,255,255,255,255,255,255,0,0,0,0,0,0,1,0,0,0,56,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,131,0,65,27,29,100,29,111,29,119,29,110,29,108,29,105,29,110,29,107,29,47,133,0,39,0,1,110,0,0,0,0,0,0,255,3,254,255,255,255,255,255,255,7,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,98,120,0,39,25,120,0,65,0>>},[deveui]}

2019-11-10 12:59:28.028 [error] <0.7472.0> connector Broker json_syntax_error**

gotthardp, 2019-11-11:
This indicated a mismatch between the "Subscribe" and the "Receieved Topic" fields. You subscribe for "in/#", the topic of the received message is "in/70B3D59BA0009871" but you match this against the "Received Topic" pattern "downlink/{deveui}", which doesn't match hence the error.

The "Received Topic" pattern should be "in/{deveui}".

mfabdul01, 2019-11-11:
hi ,
i correct my "Received Topic" pattern as "in/{deveui}" but still it showing following error,
`2019-11-11 11:52:15.464 [error] <0.24118.0> connector Broker json_syntax_error`

but the same topic name and data send from websocket is accepting

gotthardp, 2019-11-12:
And the log shows something?

mfabdul01, 2019-11-13:
[image]
the issue is the data i am sending is in Buffer format so its showing json syntax error as you can see above in the node red message of this gateway how can i convert this into Json string format?

gotthardp, 2019-11-13:
Iam confused now. What is "buffer format"?

mfabdul01, 2019-11-13:
this is the same data which i am sending to the gate way. do you aware of node-red?

gotthardp, 2019-11-15:
I know node-red, but not in very details.

mfabdul01, 2019-12-22:
The data which i am sending is a Decimal String can [check here](https://onlinestringtools.com/convert-decimal-to-string) so as it showing the error and then serialize the data through Node-Red and then its accepting the downlink

closed 2019-12-22 by mfabdul01