gotthardp/lorawan-server #685: About tx_freq_min/tx_freq_max

YonglunHo, 2019-11-12:
global_conf.json

[image]
ERROR
[image]

gotthardp, 2019-11-12:
??

YonglunHo, 2019-11-13:
tx_freq_min:510000000,tx_freq_max:0
How should I set it?
[image]

gotthardp, 2019-11-13:
In my server, the frequencies are represented by channels. If you want to constrain the frequencies, set a limited set of channels:
https://github.com/gotthardp/lorawan-server/blob/master/doc/Infrastructure.md#channels
https://github.com/gotthardp/lorawan-server/blob/master/doc/Devices.md#adaptive-data-rate-adr

YonglunHo, 2019-11-15:
Use TTN is OK.But lorawan-server Not working
TTN
[image]
lorawan-server
[image]

gotthardp, 2019-11-15:
I see. Its due to a misconfiguration.

YonglunHo, 2019-11-15:
Thank you for replying,
 I don't know my server is set up wrong.
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

YonglunHo, 2019-11-15:
I changed my Settings, but the system prompted not_semtech_mote
[image]

YonglunHo, 2019-11-18:
I set Handlers. Sometimes it works. But more often than not, it's CRC_FAIL: 100.00%
Is it because of network --> channles? I set Initial Channels 0-95
##### 2019-11-18 02:56:50 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 15
# CRC_OK: 0.00%, CRC_FAIL: 100.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 1 (114 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:2, rej:0)
# TX rejected (collision beacon): 0.00% (req:2, rej:0)
# TX rejected (too late): 0.00% (req:2, rej:0)
# TX rejected (too early): 0.00% (req:2, rej:0)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 602444852
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# GPS sync is disabled
##### END #####

closed 2019-12-05 by YonglunHo