gotthardp/lorawan-server #687: ERROR: bad argument in call createCollection

bachehkaraji, 2019-11-16:
Hi all.

```
Format: JSON
URI: mongodb://localhost:27017
Publish Uplinks: test/tst1
Publish Events:
```
Here is the error log I get
```
lorawan_connector_mongodb:init:31 Connecting mongo to mongodb localhost:27017
...
lorawan_connector_mongodb:log_termination:94 Connector mongo terminated: {badarg,[{erlang,apply,[{mongoapi,[mongo,<<"test">>]},createCollection,[<<"tst1">>]],[]},{lorawan_connector_mongodb,store_fields,3,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_mongodb.erl"},{line,112}]},{lorawan_connector_mongodb,handle_info,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_mongodb.erl"},{line,68}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,637}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,711}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}
lorawan_connector_mongodb:store_fields:112 gen_server <0.19076.2> terminated with reason: bad argument in call to erlang:apply({mongoapi,[mongo,<<"test">>]}, createCollection, [<<"tst1">>]) in lorawan_connector_mongodb:store_fields/3 line 112
lorawan_connector_mongodb:store_fields:112 CRASH REPORT Process <0.19076.2> with 0 neighbours crashed with reason: bad argument in call to erlang:apply({mongoapi,[mongo,<<"test">>]}, createCollection, [<<"tst1">>]) in lorawan_connector_mongodb:store_fields/3 line 112
lorawan_connector_mongodb:store_fields:112 Supervisor lorawan_connector_sup had child {mongodb,<<"mongo">>} started with lorawan_connector_mongodb:start_link({connector,<<"mongo">>,<<"Test">>,<<"json">>,<<"mongodb://localhost:27017">>,undefined,<<"test/tst...">>,...}) at <0.19076.2> exit with reason bad argument in call to erlang:apply({mongoapi,[mongo,<<"test">>]}, createCollection, [<<"tst1">>]) in lorawan_connector_mongodb:store_fields/3 line 112 in context child_terminated
...
```

Does Anyone Have A Solution For This Problem?

bachehkaraji, 2019-11-22:
I'm still waiting for someone to help me with this issue ...

bachehkaraji, 2019-11-26:
Is there no solution to this problem?

```
mongod --version

db version v4.2.1
git version: edf6d45851c0b9ee15548f0f847df141764a317e
OpenSSL version: OpenSSL 1.1.0l  10 Sep 2019
allocator: tcmalloc
modules: none
build environment:
    distmod: debian92
    distarch: x86_64
    target_arch: x86_64
```

```
cat /usr/lib/erlang/releases/RELEASES

%% coding: utf-8
[{release,"Erlang/OTP","22","10.5.5",
          [{kernel,"6.5","/usr/lib/erlang/lib/kernel-6.5"},
           {stdlib,"3.10","/usr/lib/erlang/lib/stdlib-3.10"},
           {sasl,"3.4.1","/usr/lib/erlang/lib/sasl-3.4.1"}],
          permanent}].
```
Please help me with this problem.

bachehkaraji, 2019-11-27:
I downgraded Erlang to v21.2.6.

```
...
lorawan_connector_mongodb:log_termination:94 Connector mongo terminated: {badarg,[{erlang,apply,[{mongoapi,[mongo,<<"test">>]},createCollection,[]],[]},{lorawan_connector_mongodb,store_fields,3,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_mongodb.erl"},{line,112}]},{lorawan_connector_mongodb,handle_info,2,[{file,"/home/travis/build/gotthardp/lorawan-server/_build/default/lib/lorawan_server/src/lorawan_connector_mongodb.erl"},{line,68}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,637}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,711}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,249}]}]}
lorawan_connector_mongodb:store_fields:112 gen_server <0.8626.0> terminated with reason: bad argument in call to erlang:apply({mongoapi,[mongo,<<"test">>]}, createCollection, []) in lorawan_connector_mongodb:store_fields/3 line 112
lorawan_connector_mongodb:store_fields:112 CRASH REPORT Process <0.8626.0> with 0 neighbours crashed with reason: bad argument in call to erlang:apply({mongoapi,[mongo,<<"test">>]}, createCollection, []) in lorawan_connector_mongodb:store_fields/3 line 112
lorawan_connector_mongodb:store_fields:112 Supervisor lorawan_connector_sup had child {mongodb,<<"mongo">>} started with lorawan_connector_mongodb:start_link({connector,<<"mongo">>,<<"Test">>,<<"json">>,<<"mongodb://localhost:27017">>,undefined,<<"test/...">>,...}) at <0.8626.0> exit with reason bad argument in call to erlang:apply({mongoapi,[mongo,<<"test">>]}, createCollection, []) in lorawan_connector_mongodb:store_fields/3 line 112 in context child_terminated
lorawan_connector_mongodb:init:31 Connecting mongo to mongodb localhost:27017
...
```

milkywayisdead, 2019-12-27:
I got the same problem. Getting some help would be great.

altishchenko, 2019-12-30:
Hello, Seasons Greetings!
First of all, can you guys share screenshots of your connector configurations?

С уважением,
Александр Тищенко
тел: +7 (916) 704-61-57
ATishch@gmail.com

JUNJUEYU, 2020-09-04:
Same to me ，who can give me some help？

altishchenko, 2020-09-04:
And the same question: "First of all, can you guys share screenshots of your connector configurations?" Can't tell you more without seeing your connector config.

altishchenko, 2020-11-02:
@milkywayisdead , @bachehkaraji , @JUNJUEYU : Guys anybody of you around?
I think I found the solution and I need you to test it - I don't have mongo access at the moment.
Any one who compiled its server from source is welcome!

And I would like to know in advance your ERL version used for compilation or, if you used pre-compiled binaries, which one.

altishchenko, 2020-11-02:
Ok, anyone who is still interested in the solution, please let me know and I'll send you a _binary_ file to replace `lorawan_connector_mongodb.beam` in the server directory to test.

closed 2020-11-15 by gotthardp

bennycyau, 2020-12-22:
> Ok, anyone who is still interested in the solution, please let me know and I'll send you a _binary_ file to replace `lorawan_connector_mongodb.beam` in the server directory to test.

Hello!
Can I still get the BEAM file to test? 

platform: version 0.6.7, OTP 23, amd64

Thanks!

altishchenko, 2020-12-23:
> > Ok, anyone who is still interested in the solution, please let me know and I'll send you a _binary_ file to replace `lorawan_connector_mongodb.beam` in the server directory to test.
> 
> Hello!
> Can I still get the BEAM file to test?
> 
> platform: version 0.6.7, OTP 23, amd64
> 
> Thanks!

https://yadi.sk/d/Mi9EpF07fwQWZg

bennycyau, 2021-01-18:
> https://yadi.sk/d/Mi9EpF07fwQWZg

Thanks for the quick response.  I tried the patch today and it worked great!  I got it to send data to both localhost and remote mongodb servers.  Actually I tried the patch on both Windows OS and Ubuntu installations, and it worked both places.

commit 7217e58c (2020-11-15): Merge pull request #757 from altishchenko/notuples

Eliminate tuple_calls in mongodb connector. Fixes #687
commit cdf2ddfe (2020-11-15): Eliminate tuple_calls in mongodb connector. Fixes #687