gotthardp/lorawan-server #700: Downlink get queued not processing immediately in C class device

mfabdul01, 2020-01-14:
Using 1GATEWAY Lora server in which i integrated a C class device to control a ON/OFF condition in Digital output of a Servo Motor. i using MQTT protocol for the communication. issue is every time i sent a Downlink it is queuing, not processing the frame immediately.
`{"data":"411001", "port":2, "time":"immediately"} `
if i connect with MQTT client MQTT. fx and send the following command,
`{"data":"411001", "port":2, "time":"immediately"} `
its processing it immediately

Apaisal, 2020-09-16:
Dear @gotthardp 

I also found this issue sometime. Downlink has been queued and wait a few minute before transmission command to specific device or never.
Question
1. Is it related about gateway duty cycle or not?
2. How to traceback which module spent network timing after registered downlink in queue?

altishchenko, 2020-09-16:
@Apaisal Hello Anol. Please be a little more specific about your setup: What version of the server are you using? And Erlang? What method/protocol do you use to send downlink? Does this situation happen when you send downlink by hand via websocket of http, for example?