gotthardp/lorawan-server #701: (master branch) Publishing AMQP events with a pattern including {devaddr} not working

citruspress, 2020-01-24:
I'm using the master branch since we can't build the Docker container on arm64 using the stable branch.

I have the following configuration:
[image]

Uplink works as expected. However I don't get any events when using the "Test" button on the handler. It's not published to `/eventAA508344` or to `/event`.

If I hard code the Publish Events field to `/eventAA508344` instead I get the events as expected. 

I am including the devaddr field with events like this in the application:
[image]

If it helps I could have a look and see if it works using the stable branch.

The debug log after clicking the test button:

`2020-01-24 12:19:54.237 [debug] <0.20145.0>@lorawan_admin_connections:handle_action:121 Sending connector test <<"LoraTestGw">> to <<"AA508344">>
2020-01-24 12:19:54.238 [debug] <0.20146.0> Supervisor {<0.20146.0>,amqp_channel_sup} started amqp_gen_consumer:start_link(amqp_selective_consumer, [], {<<"client 172.17.0.2:46535 -> 192.168.32.137:5672">>,1}) at pid <0.20147.0>
2020-01-24 12:19:54.238 [debug] <0.20146.0> Supervisor {<0.20146.0>,amqp_channel_sup} started amqp_channel:start_link(network, <0.20124.0>, 1, <0.20147.0>, {<<"client 172.17.0.2:46535 -> 192.168.32.137:5672">>,1}) at pid <0.20148.0>
2020-01-24 12:19:54.239 [debug] <0.20146.0> Supervisor {<0.20146.0>,amqp_channel_sup} started rabbit_writer:start_link(#Port<0.4178>, 1, 131072, rabbit_framing_amqp_0_9_1, <0.20148.0>, {<<"client 172.17.0.2:46535 -> 192.168.32.137:5672">>,1}) at pid <0.20149.0>`

gotthardp, 2020-01-27:
Hello. Thanks for the detailed report. Could you try again, please? Thanks to @altishchenko I fixed a bug related to publishing events via AMQP.

citruspress, 2020-02-03:
It's working as expected now. Thanks!

closed 2020-02-03 by citruspress