gotthardp/lorawan-server #709: Confirmed immediate downlink

citruspress, 2020-02-28:
When sending a confirmed immediate downlink message like the following:
'{"data":"04000000000001000000", "confirmed":true, "port": 2, "time":"immediately"}'

We never get a confirmation from the device and all following messages are just "re-up" (red arrows).

Sending unconfirmed immediate doesn't work but doesn't cause any errors on the following messages afterwards.

Sending class A messages works as expected but the device is a class C device so we prefer immediate messages.

The log file is pasted below.

2020-02-28 16:48:02.122 [info] <0.6523.1>@lorawan_gw_forwarder:terminate:131 packet_forwarder interface terminated: {function_clause,[{lorawan_gw_forwarder,build_txpk,[{txq,869.525,<<"SF9BW125">>,<<"4/5">>,immediately,16},undefined,0,<<160,51,204,187,170,0,22,0,2,251,85,206,121,45,214,22,251,28,13,236,174,121,49>>],[{file,"/lorawan-server/src/lorawan_gw_forwarder.erl"},{line,191}]},{lorawan_gw_forwarder,handle_info,2,[{file,"/lorawan-server/src/lorawan_gw_forwarder.erl"},{line,112}]},{gen_server,try_dispatch,4,[{file,"gen_server.erl"},{line,616}]},{gen_server,handle_msg,6,[{file,"gen_server.erl"},{line,686}]},{proc_lib,init_p_do_apply,3,[{file,"proc_lib.erl"},{line,247}]}]}
2020-02-28 16:48:02.122 [error] <0.6523.1>@lorawan_gw_forwarder:build_txpk:191 gen_server lorawan_gw_forwarder terminated with reason: no function clause matching lorawan_gw_forwarder:build_txpk({txq,869.525,<<"SF9BW125">>,<<"4/5">>,immediately,16}, undefined, 0, <<160,51,204,187,170,0,22,0,2,251,85,206,121,45,214,22,251,28,13,236,174,121,49>>) line 191
2020-02-28 16:48:02.123 [error] <0.6523.1>@lorawan_gw_forwarder:build_txpk:191 CRASH REPORT Process lorawan_gw_forwarder with 0 neighbours crashed with reason: no function clause matching lorawan_gw_forwarder:build_txpk({txq,869.525,<<"SF9BW125">>,<<"4/5">>,immediately,16}, undefined, 0, <<160,51,204,187,170,0,22,0,2,251,85,206,121,45,214,22,251,28,13,236,174,121,49>>) line 191
2020-02-28 16:48:02.123 [error] <0.3547.1>@lorawan_gw_forwarder:build_txpk:191 Supervisor {<0.3547.1>,lorawan_gw_sup} had child packet_forwarder started with lorawan_gw_forwarder:start_link([{port,1680}]) at <0.6523.1> exit with reason no function clause matching lorawan_gw_forwarder:build_txpk({txq,869.525,<<"SF9BW125">>,<<"4/5">>,immediately,16}, undefined, 0, <<160,51,204,187,170,0,22,0,2,251,85,206,121,45,214,22,251,28,13,236,174,121,49>>) line 191 in context child_terminated
2020-02-28 16:48:02.123 [debug] <0.3547.1> Supervisor {<0.3547.1>,lorawan_gw_sup} started lorawan_gw_forwarder:start_link([{port,1680}]) at pid <0.6839.1>
2020-02-28 16:50:47.377 [warning] <0.7008.1> node AABBCC33 downlink_missed
2020-02-28 16:50:47.578 [debug] <0.7008.1>@lorawan_handler:retransmit:236 AABBCC33 retransmitting 2
2020-02-28 16:51:03.862 [warning] <0.7016.1> node AABBCC33 downlink_missed
2020-02-28 16:51:04.063 [debug] <0.7016.1>@lorawan_handler:retransmit:236 AABBCC33 retransmitting 3
2020-02-28 16:51:20.513 [warning] <0.7027.1> node AABBCC33 downlink_missed
2020-02-28 16:51:20.714 [debug] <0.7027.1>@lorawan_handler:retransmit:236 AABBCC33 retransmitting 4
2020-02-28 16:51:54.629 [warning] <0.7038.1> node AABBCC33 downlink_missed
2020-02-28 16:51:54.831 [debug] <0.7038.1>@lorawan_handler:retransmit:236 AABBCC33 retransmitting 5
2020-02-28 16:52:39.886 [warning] <0.7053.1> node AABBCC33 downlink_missed
2020-02-28 16:52:40.087 [debug] <0.7053.1>@lorawan_handler:retransmit:236 AABBCC33 retransmitting 6
2020-02-28 16:53:25.615 [warning] <0.7091.1> node AABBCC33 downlink_missed
2020-02-28 16:53:25.816 [debug] <0.7091.1>@lorawan_handler:retransmit:236 AABBCC33 retransmitting 7

[image]

[image]

[image]

[image]

[image]

WillooWisp, 2020-03-09:
I experience the same problem and according to the log there is a crash in the forwarder. Any known issue or tip on how to solve this?

CRASH REPORT Process lorawan_gw_forwarder with 0 neighbours crashed with reason: no function clause matching lorawan_gw_forwarder:build_txpk

citruspress, 2020-03-09:
Turns out this is a separate issue. It works as expected on stable.

I thought we were running stable but since the Dockerfile clones the git repository without checking out any branches the stable image is built on the latest branch.

Maybe this is a separate discussion, but I think the Dockerfile should use COPY instead of git clone, is there a reason for cloning the repository again?

Would you be interested in a pull request changing that and looking into adding multi stage build (https://docs.docker.com/develop/develop-images/multistage-build/) to reduce the image size?

gotthardp, 2020-03-09:
Yes, pull requests are always welcome. There is no reason for not using COPY, I guess. It's just my poor Docker expertise.

WillooWisp, 2020-03-09:
Seems to work on stable but not on latest, the problem is that the stable docker container is not really running code from the stable branch, but from the master branch, pretty confusing.