gotthardp/lorawan-server #710: nucleo-lrwan2 +Lorawan-server

KVX-2018, 2020-03-03:
Hello,
i'm bought the  LORA Gateway kit from  STm, link https://www.st.com/en/evaluation-tools/p-nucleo-lrwan2.html
Downlink data recieve by server in ABP mode. But does not work OTAA. I found that issues is "does not work" uplink. nucleo-lrwan  does not see TX package from Server

LOG from nucleo-lrwan2 when LORA module does not receive confirmation. (where is not TX)
JUL: {"rxpk":[{"tmst":52689860,"chan":5,"rfch":1,"freq":868.100000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":10.3,"rssi":-45,"size":23,"data":"AEtKAPB+1bNwBgAAAAAAABBff7Ti9OE="}]}
LRX: 004B4A00F07ED5B3700600000000000010607F4E7612C3

Debug LOR from Server when LORA module does not receive confirmation . 
version of server 0.6.7

2020-03-03 19:00:26.105 [debug] <0.1568.0>@lorawan_mac:encode_accept:441 Join-Accept <<"02FDF67B">>, netid <<0,0,2>>, cflist [], rx1droff 0, rx2dr 0, appkey <<"0102030405060708090A0B0C0D0E0F10">>, appnce <<"6E76D5">>
2020-03-03 19:00:26.106 [debug] <0.1568.0>@lorawan_handler:join:114 Join-Accept in RX1: {rxq,868.3,<<"SF8BW125">>,<<"4/5">>,undefined,undefined,120115420,-38,11.3}
2020-03-03 19:00:27.883 [info] <0.346.0> [emqttc_RPI_b84c36fabd806bc6d122@192.168.0.23:43983] SENT: PINGREQ(Q0, R0, D0)
2020-03-03 19:00:27.883 [info] <0.346.0> [emqttc_RPI_b84c36fabd806bc6d122@192.168.0.23:43983] SENT: <<192,0>>
2020-03-03 19:00:52.164 [info] <0.1585.0> device 1000000000000006 {join,<<"02FDF67B">>}
2020-03-03 19:00:52.165 [warning] <0.1585.0> node 02FDF67B {repeated_reset,97}
2020-03-03 19:00:52.167 [debug] <0.1585.0>@lorawan_mac:encode_accept:441 Join-Accept <<"02FDF67B">>, netid <<0,0,2>>, cflist [], rx1droff 0, rx2dr 0, appkey <<"0102030405060708090A0B0C0D0E0F10">>, appnce <<"6CF652">>
2020-03-03 19:00:52.168 [debug] <0.1585.0>@lorawan_handler:join:117 Join-Accept in RX2: {rxq,868.1,<<"SF8BW125">>,<<"4/5">>,undefined,undefined,146157252,-43,12.0} {0,0,869.525}

Setting of Lora server
[image]

But with thethingsnetwork the nucleo-lrwan2 works well.
LOG from nucleo-lrwan2
LRX: 004B4A00F07ED5B3700600000000000010A2C13C68FF06
JUL: {"rxpk":[{"tmst":71332428,"chan":5,"rfch":1,"freq":868.100000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9.5,"rssi":-52,"size":23,"data":"AEtKAPB+1bNwBgAAAAAAABCiwTxo/wY="}]}
JDL: {"txpk":{"imme":false,"tmst":76332428,"freq":868.1,"rfch":0,"powe":11,"modu":"LORA","datr":"SF7BW125","codr":"4/5","ipol":true,"size":33,"ncrc":true,"data":"IKgHtrn/XcsfEjHBjMEFYPk0svshRjMz3uyOwK8vgq3V"}}
LTX: 20A807B6B9FF5DCB1F1231C18CC10560F934B2FB21463333DEEC8EC0AF2F82ADD5
LTS: 75611181 76332428 721247
LRX: 40B820012680000001A1166445
JUL: {"rxpk":[{"tmst":77742724,"chan":5,"rfch":1,"freq":868.100000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":10.5,"rssi":-42,"size":13,"data":"QLggASaAAAABoRZkRQ=="}]}
JDL: {"txpk":{"imme":false,"tmst":79742724,"freq":869.525,"rfch":0,"powe":24,"modu":"LORA","datr":"SF9BW125","codr":"4/5","ipol":true,"size":17,"ncrc":true,"data":"YLggASaFAAADQf8AAXT39EU="}}
LTX: 60B82001268500000341FF000174F7F445
LTS: 79071820 79742724 670904

Setting of nucleo-lrwan2

PLINK UDP PORT: 1700
 DOWNLINK UDP PORT: 1700
          CHANNEL0: 867100000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL1: 867300000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL2: 867500000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL3: 867700000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL4: 867900000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL5: 868100000, B, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL6: 868300000, B, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL7: 868500000, B, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL8: 868300000, B, SF7, BW250KHz         (LORA_STANDARD)
          CHANNEL9: 868800000, B, 50Kbps                (FSK)
-------------------------------------------------------------------------------

a link https://www.thethingsnetwork.org/docs/gateways/st/UM2587_Rev1pub.pdf

nucleo-lrwan2 used as a basic LoRaWAN® packet forwarder. In that way, data coming from the development node can reach LoRaWAN® network servers directly.

Has anyone tested nucleo-lrwan2, knows why it doesn’t work?

altishchenko, 2020-03-04:
Can you show what is in your server's sys.config file? By your output it looks like the reply from the server is not reaching your gateway by network.

KVX-2018, 2020-03-04:
Hello, Alexander

Server's sys.config file:
    % server configuration
{lorawan_server, [
    % UDP port listening for packets from the packet_forwarder Gateway
    {packet_forwarder_listen, [{port, 1680}]},
    % HTTP port for web-administration and REST API
    {http_admin_listen, [{port, 8080}]},
    % HTTP/SSL port
    {http_admin_listen_ssl, [
        {port, 8443},
        {certfile, "cert.pem"},
        {cacertfile, "cacert.pem"},
        {keyfile, "key.pem"}

NW Server  and GW there is in locall network
Setting of GW
AT+PKTFWD
+PKTFWD: 192.168.0.29, 1680, 1680

On RPI+Ic880 all work well

RPI+Ic880 settings:

  "gateway_conf": {
        "gateway_ID": "B827EBFFFECB5231",
        /* change with default server address/ports, or overwrite in local_conf.json */
        "server_address": "localhost",
        "serv_port_up": 1680,
        "serv_port_down": 1680,

altishchenko, 2020-03-04:
Yep. but your stm gateway config uses port 1700 instead of 1680. Try changing either server or gateway to match - 1700 on both or 1680 on both

KVX-2018, 2020-03-04:
1700 - is setting for TTN Server, to check that works.

But maybe set 1700 on Server. Just try.

So setting of port 1700 instead of 1680   is under control

04.03.2020 14:19, Alexander Tishchenko пишет:

altishchenko, 2020-03-04:
yes, but restart the server after that

KVX-2018, 2020-03-04:
Changed

{lorawan_server, [
     % UDP port listening for packets from the packet_forwarder Gateway
     {packet_forwarder_listen, [{port, 1700}]},

AT+PKTFWD
+PKTFWD: 192.168.0.29, 1700, 1700

Situation is full the same. No  uplink

04.03.2020 14:26, Alexander Tishchenko пишет:

altishchenko, 2020-03-04:
Strange enough. Can you confirm that UDP data exchange between gw and the server really happens both ways? Depending on your OS you can run on **your server** wireshark (windows) or tcpdump (unix/linux) to see the UDP packets on port 1700. Linux ex.: `tcpdump -ieth0 udp port 1700`.
Or do you by any chance run under docker? If so, check the docker port mapping configuration too!

KVX-2018, 2020-03-04:
Capture log in attached, possible open by wireshark.

I think server send data to GW to port 1700

maybe here issue.

altishchenko, 2020-03-04:
sorry, you forgot the log :)

KVX-2018, 2020-03-04:
[
{
"@context": "http://schema.org",
"@type": "EmailMessage",
"potentialAction": {
"@type": "ViewAction",
"target": "https://github.com/gotthardp/lorawan-server/issues/710?email_source=notifications\u0026email_token=AKDIS7HQDI57ULTHIIK24E3RF2DN5A5CNFSM4LARSNWKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOENY65DQ#issuecomment-594669198",
"url": "https://github.com/gotthardp/lorawan-server/issues/710?email_source=notifications\u0026email_token=AKDIS7HQDI57ULTHIIK24E3RF2DN5A5CNFSM4LARSNWKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOENY65DQ#issuecomment-594669198",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

KVX-2018, 2020-03-10:
Made capture the gotthardp and TTN packets from server to GW.
It can be seen that the order of the fields is different.. TTN use order field as it is  Semtech package forwarder.  It is possible for the STm SW of GW takes strong order of fields as it there in Semtech package forwarder. 

TTN
{"txpk":[{"imme":false,"tmst":529340084,"freq":869.525,"rfch":0,"powe":24,"modu":"LORA","datr":"SF12BW125","codr":"4/5","ipol":true,"size":33,"ncrc":true,"data":"IF4KZ7kT/oZWNMVeFODWnh7Cz5S3xcKEua4SrCtf26gx"}}

Server gotthardp
{"txpk":{"powe":3,"imme":false,"tmst":754711964,"codr":"4/5","datr":"SF12BW125","freq":869.525,"modu":"LORA","rfch":0,"ipol":true,"size":17,"data":"IKNiacwglkONa4KPnTsJ5QI="}}

Order of fields  of Semtech package forwarder
{"txpk":{
	"imme":true,
	"freq":864.123456,
	"rfch":0,
	"powe":14,
	"modu":"LORA",
	"datr":"SF11BW125",
	"codr":"4/6",
	"ipol":false,
	"size":32,
	"data":"H3P3N2i9qc4yt7rK7ldqoeCVJGBybzPY5h1Dd7P7p8v"
}}

altishchenko, 2020-03-10:
The fields order is irrelevant, because it is JSON. What I was suspecting is, that the communication ports between the server and the gateway mismatch, may be firewall or docker issue, if you have them. You don't see these errors with TTN, because TTN is considered by your network/router as an external Internet resource. I'll have a look into your wireshark as soon as you approve Google disk access request.

KVX-2018, 2020-03-10:
The fields order is irrelevant, because it is JSON.
- YES, but maybe it is issue of STm of SW.

may be firewall or docker issue
  - does not use docker. RX packages receive by gotthardp and definitely 
server send TX packages  to STM GW  but STM GW  does not recognise it.

As STm of GW  as TTN as gotthardp server use Semtech packet forwarder.
One point that i fond it is different order of JSON fields.

No more ideas why STM GW does not work with gotthardp server.

altishchenko, 2020-03-12:
I re-requested the access to you wireshark capture from address atishch@..., please confirm it or make the capture shareable by link. I'd like to have a closer look at what is going on in the packet exchange.

mitwave, 2020-03-14:
Hi.
Let’s try to enable the log on GW.
AT+LOG=1
You can find the Ethernet packet log from GW via USB virtual UART.

I’m using ST GW with this server.
It’s working well

KVX-2018, 2020-03-17:
Hello mitwave,
What version of FW you use?

LOG

           VERSION: 2.1.7, Nov  6 2018
               LOG: ON
           AT ECHO: ON
          BAUDRATE: 115200bps
           MACADDR: B8:27:EB:CB:52:32
          ETHERNET: DHCP
              DNS1: 192.168.0.1
              DNS2: 192.168.0.1
        NTP SERVER: 1.ubuntu.pool.ntp.org
       EUI PADDING: {3, FF}, {4, FE}
        GATEWAY ID: B827EBFFFECB5232
           LORAWAN: Public
    LORAWAN SERVER: 192.168.0.29
   UPLINK UDP PORT: 1700
 DOWNLINK UDP PORT: 1700
          CHANNEL0: 867100000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL1: 867300000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL2: 867500000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL3: 867700000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL4: 867900000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL5: 868100000, B, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL6: 868300000, B, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL7: 868500000, B, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL8: 868300000, B, SF7, BW250KHz         (LORA_STANDARD)
          CHANNEL9: 868800000, B, 50Kbps                (FSK)
-------------------------------------------------------------------------------
Concentrator starting...
Concentrator Radio A type SX1257
Concentrator Radio B type SX1257
Radio A, min:866950000, max:868050000
Radio B, min:867950000, max:868862500
Concentrator started (2926ms)
ST LoRa GW V2
Ethernet starting...

.............
LRX: 004B4A00F07ED5B3700600000000000010B491DF9E4FAA
JUL: {"rxpk":[{"tmst":50643428,"chan":6,"rfch":1,"freq":868.300000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9.0,"rssi":-40,"size":23,"data":"AEtKAPB+1bNwBgAAAAAAABC0kd+eT6o="}]}
LRX: 004B4A00F07ED5B3700600000000000010B5916B367759
JUL: {"rxpk":[{"tmst":56902268,"chan":6,"rfch":1,"freq":868.300000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9.0,"rssi":-34,"size":23,"data":"AEtKAPB+1bNwBgAAAAAAABC1kWs2d1k="}]}
LRX: 004B4A00F07ED5B3700600000000000010B691BD35A226
JUL: {"rxpk":[{"tmst":63212892,"chan":5,"rfch":1,"freq":868.100000,"stat":1,"modu":"LORA","datr":"SF8BW125","codr":"4/5","lsnr":11.3,"rssi":-39,"size":23,"data":"AEtKAPB+1bNwBgAAAAAAABC2kb01oiY="}]}
LRX: 004B4A00F07ED5B3700600000000000010B7917E890763
JUL: {"rxpk":[{"tmst":83276588,"chan":6,"rfch":1,"freq":868.300000,"stat":1,"modu":"LORA","datr":"SF8BW125","codr":"4/5","lsnr":11.5,"rssi":-32,"size":23,"data":"AEtKAPB+1bNwBgAAAAAAABC3kX6JB2M="}]}

And LOG with TTN Server
        VERSION: 2.1.7, Nov  6 2018
               LOG: ON
           AT ECHO: ON
          BAUDRATE: 115200bps
           MACADDR: B8:27:EB:CB:52:32
          ETHERNET: DHCP
              DNS1: 192.168.0.1
              DNS2: 192.168.0.1
        NTP SERVER: 1.ubuntu.pool.ntp.org
       EUI PADDING: {3, FF}, {4, FE}
        GATEWAY ID: B827EBFFFECB5232
           LORAWAN: Public
    LORAWAN SERVER: router.eu.thethings.network
   UPLINK UDP PORT: 1700
 DOWNLINK UDP PORT: 1700
          CHANNEL0: 867100000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL1: 867300000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL2: 867500000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL3: 867700000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL4: 867900000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL5: 868100000, B, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL6: 868300000, B, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL7: 868500000, B, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL8: 868300000, B, SF7, BW250KHz         (LORA_STANDARD)
          CHANNEL9: 868800000, B, 50Kbps                (FSK)
-------------------------------------------------------------------------------
Concentrator starting...
Concentrator Radio A type SX1257
Concentrator Radio B type SX1257
Radio A, min:866950000, max:868050000
Radio B, min:867950000, max:868862500
Concentrator started (2926ms)
ST LoRa GW V2
Ethernet starting...
Ethernet started
DHCP IP: 192.168.0.91
Downlink UDP Connected
Uplink UDP Connected
LRX: 004B4A00F07ED5B370060000000000001012B2632F41B8
JUL: {"rxpk":[{"tmst":32130564,"chan":5,"rfch":1,"freq":868.100000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9.0,"rssi":-35,"size":23,"data":"AEtKAPB+1bNwBgAAAAAAABASsmMvQbg="}]}
JDL: {"txpk":{"imme":false,"tmst":37130564,"freq":868.1,"rfch":0,"powe":11,"modu":"LORA","datr":"SF7BW125","codr":"4/5","ipol":true,"size":33,"ncrc":true,"data":"INMBX3WtLjgvrPXt+E9PRf6RHG4INn3xu1A046XnowSb"}}
LTX: 20D3015F75AD2E382FACF5EDF84F4F45FE911C6E08367DF1BB5034E3A5E7A3049B
LTS: 36400227 37130564 730337
LRX: 40A324012680000001B7A5F86C
JUL: {"rxpk":[{"tmst":38541364,"chan":6,"rfch":1,"freq":868.300000,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","lsnr":10.0,"rssi":-39,"size":13,"data":"QKMkASaAAAABt6X4bA=="}]}
JDL: {"txpk":{"imme":false,"tmst":40541364,"freq":869.525,"rfch":0,"powe":24,"modu":"LORA","datr":"SF9BW125","codr":"4/5","ipol":true,"size":17,"ncrc":true,"data":"YKMkASaFAAADQf8AAUdotEc="}}
LTX: 60A32401268500000341FF00014768B447
LTS: 39865059 40541364 676305

mitwave, 2020-03-18:
Hi.
When GW try to connect the internet, GW can’t get the IP from DHCP server.
When you find this message in the log, this is Ok.

Ethernet starting...
Ethernet started
DHCP IP: 192.168.0.91
Downlink UDP Connected
Uplink UDP Connected

But when you face the problem, GW log is only

Ethernet starting...

So, you should check the network of LAN.
I guess, there is no DHCP server in your local network...

KVX-2018, 2020-03-18:
Hi, 
i think here is OK.

LOG with IP
-------------------------------------------------------------------------------
           VERSION: 2.1.7, Nov  6 2018
               LOG: ON
           AT ECHO: ON
          BAUDRATE: 115200bps
           MACADDR: B8:27:EB:CB:52:32
          ETHERNET: DHCP
              DNS1: 192.168.0.1
              DNS2: 192.168.0.1
        NTP SERVER: 1.ubuntu.pool.ntp.org
       EUI PADDING: {3, FF}, {4, FE}
        GATEWAY ID: B827EBFFFECB5232
           LORAWAN: Public
    LORAWAN SERVER: 192.168.0.29
   UPLINK UDP PORT: 1700
 DOWNLINK UDP PORT: 1700
          CHANNEL0: 867100000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL1: 867300000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL2: 867500000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL3: 867700000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL4: 867900000, A, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL5: 868100000, B, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL6: 868300000, B, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL7: 868500000, B, SF7/SF12, BW125KHz    (LORA_MULTI_SF)
          CHANNEL8: 868300000, B, SF7, BW250KHz         (LORA_STANDARD)
          CHANNEL9: 868800000, B, 50Kbps                (FSK)
-------------------------------------------------------------------------------
Concentrator starting...
Concentrator Radio A type SX1257
Concentrator Radio B type SX1257
Radio A, min:866950000, max:868050000
Radio B, min:867950000, max:868862500
Concentrator started (2926ms)
ST LoRa GW V2
Ethernet starting...
Ethernet started
DHCP IP: 192.168.0.91
Downlink UDP Connected
Uplink UDP Connected

And server receive message,
[image]

And transmit package to IP and Port but GW  does not recognize  it , i think.
GW and this Server in the same Network

Dont understand why. 
If you say GW works with this Server, i'm confused.

mitwave, 2020-03-18:
At 1st, I want to confirm...
1. Can you see the status of GW in the Server Web console?
2. Uplink = from device to server and Downlink = from server to device. Is this OK?
3. Can you see the device Join status on Server Web Console?(I guess “Yes”)
4. What do you want to do with ST Ge and this server?

KVX-2018, 2020-03-18:
1. Can you see the status of GW in the Server Web console?
YES
2. Uplink = from device to server and Downlink = from server to device. Is this OK?
YES
3. Can you see the device Join status on Server Web Console?(I guess “Yes”)
YES
4. What do you want to do with ST Ge and this server?
GW and LORAWAN Network

mitwave, 2020-03-18:
Ok.
Now, your device could join the LoRaWAN network and you can receive the data from device vi ST GW.

What is your problem?
Downlink?
Would you explain again?

KVX-2018, 2020-03-18:
Yes, GW  does not recognize Downlink 
What FW version for GW do you use?

altishchenko, 2020-04-21:
@KVX-2018 Can you configure your LoRa node to attempt Join at SF12 instead of SF7? I had an issue with another kit and another LoRa stack library with SF7 not joining.

mrerdem, 2021-04-29:
I am in the exact same situation as KVX-2018. There is no packets from lorawan-server to the gateway (checked using tcpdump). That's why, the node is shown as joined on the server's web UI, while the node is not aware that it has joined.