gotthardp/lorawan-server #711: Gateway parameters + MQTT

javobarra, 2020-03-16:
Hi Petr! Is it possible to make independent mqtt message with Gateway parameters?? Or the only way to send Gateway parameters is when receive a node message?
i created a handler and connector for send via mqtt only Gateway parameters. But i think this is not possible to do it

Thanks

altishchenko, 2020-05-13:
@javobarra Can you please be a little more specific, on what type of information about the gateway you want to receive? Some things are possible and some not.

javobarra, 2020-05-13:
Hi @altishchenko!  
You know this kind of package. 

"rxpk":
[
{
"time":"2013-03-31T16:21:17.528002Z",
"tmst":3512348611,
"chan":2,
"rfch":0,
"freq":866.349812,
"stat":1,
"modu":"LORA",
"datr":"SF7BW125",
"codr":"4/6",
"rssi":-35,
"lsnr":5.1,
"size":32,
"data":"-DS4CGaDCdG+48eJNM3Vai-zDpsR71Pn9CPA9uCON84"
},
{
"time":"2013-03-31T16:21:17.532038Z",
"tmst":3316387610,
"chan":0,
"rfch":0,
"freq":863.00981,
"stat":1,
"modu":"LORA",
"datr":"SF10BW125",
"codr":"4/7",
"rssi":-38,
"lsnr":5.5,
"size":32,
"data":"ysgRl452xNLep9S1NTIg2lomKDxUgn3DJ7DE+b00Ass"
}
]

I want to store some of this values. But first i have to send via MQTT. My problem is if i have a method to take from the lorawan-server those values. Like a tag in "uplink fields" in Handler config, is it possible?

Thanks for your help!

altishchenko, 2020-05-13:
@javobarra Yes, you get most of these fields with 'Uplinks fields' in a Handler, they are even named the same. Just click on the empty 'Uplink fields' box to see the drop down list of them. Or do I misunderstand something?
[image]

altishchenko, 2020-05-13:
I re-read your first question once again. You are correct, you can get this data only with the message from the node, because all this data corresponds only to this message received from the node, some of these parameters will be different for other nodes and other messages of this node.

javobarra, 2020-05-14:
> I re-read your first question once again. You are correct, you can get this data only with the message from the node, because all this data corresponds only to this message received from the node, some of these parameters will be different for other nodes and other messages of this node.

Yes, you are right. 
Thanks!

closed 2020-05-14 by javobarra