gotthardp/lorawan-server #713: Can someone explain how to send a downlink

hjojohan, 2020-03-25:
Hi 
Can someone explain how to send a downlink, i want to control a relay on a class C device.  Today we use the lorawanserver only to recieve sensordata over HTTP. 

Is there a rest API we can use to send a downlink from our software ?   Is there someone who has a Python or .NET code snippet to share that could clarify the procedure for me ? 

Could we use postman to do test ?

All help much appreciated
/ Johan

altishchenko, 2020-03-25:
It is very easy :)
Create a connector with:

Application: AppYouUse
Format: JSON
URI: **http:**
Received Topic: **/in/{devaddr}**

Bold fields are what is important here. Other fields just leave blank/default.

Once this connector is enabled, you will be able to post data to you device at address `http://yourserver:8080/in/12345678", where 12345678 your device's DevAddr.

Here is a typical command:
`curl -v --digest -uadmin:admin -H "Content-Type: application/json" -X POST http://localhost:8080/in/11223344 -d "{\"data\":\"ABCDEF012345\", \"port\":2, \"time\":\"immediately\"}"`

where `port` - is the LoRaWAN port you are going to receive data, `data` is HEX encoded arbitrary binary and `immediately` tells the server it is a Class C device, don't wait for UpLink.

Make sure you read [Handlers](/gotthardp/lorawan-server/blob/master/doc/Handlers.md) and [Connectors](/gotthardp/lorawan-server/blob/master/doc/Connectors.md) in the docs.

altishchenko, 2020-04-15:
Did you manage to send the downlink as described?

sudocoffie, 2020-04-21:
Hello @altishchenko !

We can se the downlink in "Frames" but it dosenot seem to transmit to the device. The downlinks dosent display in the node either. Do you have any idea where i should look for an error?

altishchenko, 2020-04-21:
@sudocoffie Have a close look into your debug.log file while sending a downlink, also note, some devices are 'picky' on destination port, so it must be the one your device uses for communication. Downlinks are sent either on the last known frequency the device used or on a default channel. To make sure, try sending confirmed downlink - you will see, if your device acknowledges reception.

sudocoffie, 2020-04-21:
Okey, iam sure we use the correct destination port. We will try to send a confirmed downlink to se if it acks. Thanks for the speedy response!

altishchenko, 2020-04-21:
By the way, what end device do you use?

hjojohan, 2020-04-21:
Mcf88 310
https://www.mcf88.it/prodotto/mcf-lw13io/

[GPSLogik_email_signatur]
GPS Logik AB
Johan Ansgariusson Stenström
CEO / Customer Success Manager
Skövde Science Park |Kaplansgatan 16B | 541 34  Skövde | Sweden.
Cell: +46 (0)733 165 100
Tel : +46(0)500-6000 22
johan@gpslogik.com<mailto:johan@gpslogik.com?subject=GPSlogik.se> | http://www.gpslogik.se<http://www.gpslogik.se/>

GPS Logik AB har under +10 år arbetat med GPS-positioneringssystem anpassade
för den svenska marknadens behov. Våra fleetmanagementsystem ger dig alltid svar på frågor
som innehåller ord som –Vem, -Vart, -När och Hur.

Vi erbjuder världsledande produkter och system från dessa leverantörer.
www.ruptela.com<http://www.ruptela.com/>
www.digitalmatter.com<http://www.digitalmatter.com/>
www.wlius.com<http://www.wlius.com/>
www.inosat.com<http://www.inosat.com/>

altishchenko, 2020-04-21:
@hjojohan Surprisingly, device's documentation doesn't mention the port it is accepting downlink frames on. I had a look into this file: https://intra.orisun-iot.com/file/download/qjdgwvsexuy6jpw4e2aj/PHID-FILE-7fmo7szkgdg7iuhbv4tc/MCF88_Data_Frame_Format_1.17.pdf

hjojohan, 2020-04-21:
We tried port1 using TTN and got the relay to switch many times.  Now Franco there chief engineer says
[cid:image001.png@01D617D7.BDC6ACD0]

Screenshoot from TTN

[cid:image002.png@01D617D7.BDC6ACD0]

[GPSLogik_email_signatur]
GPS Logik AB
Johan Ansgariusson Stenström
CEO / Customer Success Manager
Skövde Science Park |Kaplansgatan 16B | 541 34  Skövde | Sweden.
Cell: +46 (0)733 165 100
Tel : +46(0)500-6000 22
johan@gpslogik.com<mailto:johan@gpslogik.com?subject=GPSlogik.se> | http://www.gpslogik.se<http://www.gpslogik.se/>

GPS Logik AB har under +10 år arbetat med GPS-positioneringssystem anpassade
för den svenska marknadens behov. Våra fleetmanagementsystem ger dig alltid svar på frågor
som innehåller ord som –Vem, -Vart, -När och Hur.

Vi erbjuder världsledande produkter och system från dessa leverantörer.
www.ruptela.com<http://www.ruptela.com/>
www.digitalmatter.com<http://www.digitalmatter.com/>
www.wlius.com<http://www.wlius.com/>
www.inosat.com<http://www.inosat.com/>

altishchenko, 2020-04-21:
Screenshots didn't make it into your e-mail. Can you publish them via GitHub comment?

hjojohan, 2020-04-21:
Send me your email instead, 100% easier

[GPSLogik_email_signatur]
GPS Logik AB
Johan Ansgariusson Stenström
CEO / Customer Success Manager
Skövde Science Park |Kaplansgatan 16B | 541 34  Skövde | Sweden.
Cell: +46 (0)733 165 100
Tel : +46(0)500-6000 22
johan@gpslogik.com<mailto:johan@gpslogik.com?subject=GPSlogik.se> | http://www.gpslogik.se<http://www.gpslogik.se/>

GPS Logik AB har under +10 år arbetat med GPS-positioneringssystem anpassade
för den svenska marknadens behov. Våra fleetmanagementsystem ger dig alltid svar på frågor
som innehåller ord som –Vem, -Vart, -När och Hur.

Vi erbjuder världsledande produkter och system från dessa leverantörer.
www.ruptela.com<http://www.ruptela.com/>
www.digitalmatter.com<http://www.digitalmatter.com/>
www.wlius.com<http://www.wlius.com/>
www.inosat.com<http://www.inosat.com/>

altishchenko, 2020-04-21:
atishch@gmail.com

gotthardp, 2020-05-08:
Solved. Incorrect RX2 frequency was configured.

closed 2020-05-08 by gotthardp