gotthardp/lorawan-server #714: Adding a Handler to change field names

ahiregange, 2020-04-14:
Hello,

I am trying to add a Handler which will select a few field names and change them. I do not need to interpret the PDU binary data itself.
I tried to write a fun script using the examples but I am a beginner and I may have made a mistake.
I wrote the attached script and tried to save it as ASCII, Binary and the other options too but always get a red banner saying "State Change Error".
In the Connector I will be sending the data in JSON format.

Also I am not sure if the script I wrote will do the intended purpose.

[image]

Thanks,
Arun

altishchenko, 2020-04-14:
Hi! What version of the server you are using? Have not seen this one before for a long time,

altishchenko, 2020-04-14:
The reason you are getting this error is that the page you were working with while writing the handler was open for too long and some connector or device, attached to this handler changed the handlers state. This is a very unlikely situation though. The only way out of it is to copy your fun to some buffer, like note pad. Close the Edit Handler page clicking on something else on the left or, better yet, on the `Handlers` submenu on the left.
Then you can reopen your handler, copy paste your fun into it and leaving it with `Custom Binary` selected try saving it again.
Also I'd advise, if you have any `connectors` attached to this handler, disable them before you try editing once again.

altishchenko, 2020-04-14:
About your script. It will work if you will replace all capital letter keys in the second map with some special syntax. Erlang treats words starting with a capital letter as a variable, not a key identifier.

Here are examples, on how you should change your fun() and be careful about the spaces between >> << and <=, =>:
```
#{ <<"SeqNo">> => F1, <<"FREQ">> => F2, <<"SNR">> => F3, <<"RSSI">> => F4, <<"PDU">> => F5, <<"EUI">> => F6}
```

ahiregange, 2020-04-15:
Hi Alexander,

1. Thanks! The "State Change Error" went away as soon as I made the syntax change you suggested above and I was able to save the configuration as shown here.

[image]

2. However now I get parse errors as shown.
[image]
I tried with and without the Dummy:88 (my payload is 11 bytes always so I thought maybe I need to provide some handling for it even if dummy).

Thanks,
Arun

altishchenko, 2020-04-15:
You may try putting `_Dummy` instead of `<<Dummy:88>>`. 
`_` in front of a variable name means 'don't care, so for that you don't even need << and >>

ahiregange, 2020-04-15:
Hi Alexander,

Thanks for your quick answer but I am facing a new problem before I can try this. I was doing some cleanup and renaming the Handler & Connector I had with better names and now I found that the new name is not showing up in the pulldown for Application in the Profile.
So I just deleted the Profile and recreated it with same name and got the pulldown option for Application properly.

But now I see that all the Nodes are no longer connected to that Profile (even though re-created with same name) and as I have a lot of nodes I do not want to delete them all and re-create one-by-one.

Sorry for the silly situation, but I am clearly doing something wrong.

Thanks,
Arun

ahiregange, 2020-04-15:
Hi Alexander,

I just deleted all existing Nodes and recreated and now I do not get the parse_failed errors any more!
I will followup next and see if the data is reaching our cloud.

Thanks,
Arun

ahiregange, 2020-04-15:
Hi Alexander,

I am using a test server and found that this is how the data is coming through:

{"EUI":"\u0010\u0000\u0000\b","FREQ":903.0,"PDU":"\u0000'b\u0000\u0000\b\u0001\u0000\u0000\u0000\u0000","RSSI":-66,"SNR":11.0,"SeqNo":160}

A couple of things seem wrong:
1) The field names have quotes around them which I was not expecting
2) The EUI is devaddr and PDU is data and both are not coming through as expected e.g. I expected devaddr/EUI to be of the form 10000109 or 10000005 etc.

On the plus side there are no HTTP errors being reported on Dashboard Events.

Thanks,
Arun

ahiregange, 2020-04-15:
Hi Alexander,

If I do not do uplink parsing (as shown in the image) then the data comes through beautifully at the test server:
{"data":"03013901025E00A8000000","devaddr":"10000008","fcnt":219,"freq":903.5,"lsnr":9.2,"rssi":-58}

[image]

Thanks,
Arun

altishchenko, 2020-04-15:
Quotes around names - is a JSON standard.
Server doesn't know what is inside those fields you changed, but it does know standard fields like `data` and `devaddr` and applies "conversion" to them. You may try doing this with just one field right inside the map and see how things change with Custom Binary:
```
<<"EUI">> => lorawan_utils:binary_to_hex(F6)
```

ahiregange, 2020-04-15:
Hi Alexander,

With this change there are parsing errors in the Handler:

[image]

Thanks,
Arun

altishchenko, 2020-04-15:
Can you publish your debug.log file?

altishchenko, 2020-04-15:
Just to confirm. Here is my bit of test code in the Handler Parse uplink:
```
fun(#{ devaddr := EUI } = Fields, _) ->
  Fields#{ <<"EUI">> => lorawan_utils:binary_to_hex(EUI) }
end.
```

And here is a resulting JSON part:
```
{ ..., "EUI":"040A4550"}
```

ahiregange, 2020-04-15:
Hi Alexander,

Thanks a lot!

This works. And then I added field-by-field and it still worked and then I removed Fields so that original fields do not repeat. So far so good.
Is the script sensitive to spaces being in the right place?

Thanks again,
Arun

closed 2020-04-15 by ahiregange